react-native-camera/react-native-camera #1550: [AVCaptureMovieFileOutput availableVideoCodecTypes]: unrecognized selector sent to instance

Friendly-Robot, 2018-05-08:
### Which implementation are you using
*RNCamera*

### Steps to reproduce
I'm really not doing anything fancy. The video recording works perfectly fine on Android but it crashes after `stopRecording()` on iOS.
```
        <RNCamera
          autoFocus={RNCamera.Constants.AutoFocus.on}
          captureAudio={true}
          flashMode={RNCamera.Constants.FlashMode.off}
          notAuthorizedView={<View/>}
          onCameraReady={() => this.handleCameraReady()}
          pendingAuthorizationView={<View/>}
          permissionDialogTitle={'Permission to use camera'}
          permissionDialogMessage={'We need your permission to use your camera phone'}
          ratio="16:9"
          ref={cam => this.camera = cam}
          style={styles.preview}
          type={videoFacing === 'back' ? RNCamera.Constants.Type.back : RNCamera.Constants.Type.front }
          // useCamera2Api={true}
        />
```

### Expected behaviour
Finish recording and return the `uri`.

### Actual behaviour
App crashes with the following error:
```
[AVCaptureMovieFileOutput availableVideoCodecTypes]: unrecognized selector sent to instance 0x14fb4e20
2018-05-08 00:19:49.083 Undercover[620:69647] *** Terminating app due to uncaught exception 
'NSInvalidArgumentException', reason: '-[AVCaptureMovieFileOutput availableVideoCodecTypes]: 
unrecognized selector sent to instance 0x14fb4e20'
```

### Environment
- **Node.js version**: 6.11.0
- **React Native version**: 0.54.0
- **React Native platform + platform version**: iOS 9.3.3

### react-native-camera
**Version**: "master"

jgfidelis, 2018-05-08:
Can you ry on some older version like rncamera 1.1.0?

Friendly-Robot, 2018-05-08:
Trying an older version didn't work :(
But the application runs fine on an updated iOS firmware of version 11.3. It just doesn't work for 9.3 which I guess is fine. It was the only device I had lying around to test on at the time. 
Thanks for your help anyways, @jgfidelis !

jgfidelis, 2018-05-08:
https://developer.apple.com/documentation/avfoundation/avcapturemoviefileoutput/1648381-availablevideocodectypes?language=objc

availablevideocodectypes is iOS 10.0+... this is not ideal but explains a lot.

benjaminreid, 2018-06-13:
I’ve ran into this same issue today on iOS 9.3.5. Does this mean iOS 9 is simply not supported by `react-native-camera` or is there a way to work around?

I see you can specify codes but that isn’t applicable to iOS 9 anyway according to https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#recordasyncoptions-promise

sibelius, 2018-06-13:
@benjaminreid we would a PR that make it compatible with ios9 as well

benjaminreid, 2018-06-13:
@sibelius Sorry, I don’t quite understand what you meant. Are you saying this library is _not_ compatible with iOS9?

n1ru4l, 2018-06-13:
Oh now I see, https://github.com/react-native-community/react-native-camera/blob/4fdd548611b4d4fabf6008aaea94c8f69ed9a88f/ios/RN/RNCamera.m#L454

is the culprit!

benjaminreid, 2018-06-13:
@n1ru4l I’m not specifying a codec if that helps which is what I think the piece of code you...

```js
this.camera
  .recordAsync({
    maxDuration: 10,
    quality: RNCamera.Constants.VideoQuality["720p"],
  })
```

This is the stack trace of the error...

```
Fatal Exception: NSInvalidArgumentException
0  CoreFoundation                 0x220c391b __exceptionPreprocess
1  libobjc.A.dylib                0x2185ee17 objc_exception_throw
2  CoreFoundation                 0x220c92b5 __methodDescriptionForSelector
3  CoreFoundation                 0x220c6ee1 ___forwarding___
4  CoreFoundation                 0x21ff2238 _CF_forwarding_prep_0
5  StemX                          0x2c267b -[RNCamera captureOutput:didFinishRecordingToOutputFileAtURL:fromConnections:error:]
6  AVFoundation                   0x27a714a1 __104-[AVCaptureMovieFileOutput handleDidStopRecordingNotificationForWrapper:withPayload:demoof:addMetadata:]_block_invoke219
7  AVFoundation                   0x27a98cb3 AVPerformDelegateCallbackSynchronouslyForDelegateStorageIfCurrentDelegateQueueIsQueueElseDispatchToCurrentDelegateQueue
8  AVFoundation                   0x27a98d27 __AVPerformDelegateCallbackSynchronouslyForDelegateStorageIfCurrentDelegateQueueIsQueueElseDispatchToCurrentDelegateQueue_block_invoke
9  libdispatch.dylib              0x21c31823 _dispatch_call_block_and_release
10 libdispatch.dylib              0x21c3180f _dispatch_client_callout
11 libdispatch.dylib              0x21c3fba9 _dispatch_main_queue_callback_4CF$VARIANT$mp
12 CoreFoundation                 0x22085b6d __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__
13 CoreFoundation                 0x22084067 __CFRunLoopRun
14 CoreFoundation                 0x21fd3229 CFRunLoopRunSpecific
15 CoreFoundation                 0x21fd3015 CFRunLoopRunInMode
16 GraphicsServices               0x235c3ac9 GSEventRunModal
17 UIKit                          0x266a7189 UIApplicationMain
18 StemX                          0x797f3 main (main.m:14)
19 (Missing)                      0x21c7b873 (Missing)
```

n1ru4l, 2018-06-13:
@benjaminreid can you please test https://github.com/react-native-community/react-native-camera/pull/1633

PR #1633 (2018-06-13): fix(rn-camera): fix codec backwards compat
Attempt to fix #1550

Can someone with an iOS 9 device test this?
closed 2018-06-13 by jgfidelis