react-native-camera/react-native-camera #1551: Inverted QR code is impossible to scan on Android

jonathanort, 2018-05-08:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Add component
       ```<Camera
          onBarCodeRead={this.onBarCodeRead}
       />```

2. Try scanning this code using an Android device
[image]

### Does it work with Expo Camera?
No

### Expected behaviour
The QR code should be scanned and display the correct result

### Actual behaviour
The QR code is not recognized

### Environment

    "react-native": "0.55.3",
    "react-native-camera": "^1.1.2",
     Android 7.0, Android 8.0

### react-native-camera
**Version**: npm version

jgfidelis, 2018-05-08:
if you are using <Camera>, then you are using RCTCamera and not RNCamera, please try RNCamera.

jonathanort, 2018-05-08:
@jgfidelis We're importing the RNCameraPackage in our `MainApplication.java`, so we're already using RNCamera

sibelius, 2018-05-08:
you need to use <RNCamera /> instead of <Camera />

SeamusY, 2018-07-05:
@sibelius 

Attempted that: with the following code (The camera is appearing but scanning isn't working)

```<RNCamera
          barCodeTypes={[RNCamera.Constants.BarCodeType.qr]}
          flashMode={RNCamera.Constants.FlashMode.on}
          style={styles.preview}
          onBarCodeRead={this.onBarCodeRead}
          ref={cam => (this.camera = cam)}
        >
          <Text
            style={{
              backgroundColor: "white"
            }}
          >
            {this.state.qrcode}
          </Text>
        </RNCamera>
```
Currently with an android 6.0.1 and an SDK of 27. The camera is on and off focus and never scans.

arghya95, 2018-07-05:
I have the same issue.I am also using RNCamera.

Sjiep, 2018-11-05:
@jonathanort I added this feature and it got merged into master (#1896). Because it uses the google vision API it is currently Android only.

[See docs.](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#android-googlevisionbarcodemode)

closed 2018-11-05 by Sjiep

LightningXCE, 2019-04-17:
@sjiep, thanks for the progress on getting this implemented into Android.

Is there any hope for iOS to also have this feature? Working on a project currently that desperately needs this functionality.

Thanks!

Sjiep, 2019-04-17:
@LightningXCE, as I mentioned I implemented it on the Google Vision API side which back then was only for Android. 

Going to the Google Vision [documentation](https://cloud.google.com/vision/docs/) it might be possible using the ML Kit  for Firebase, but personally I know nothing about that SDK. 

I also don't own or work with apple devices or have developed something for iOS. So I'm sorry I can't help you there.

sibelius, 2019-04-17:
try on latest version 

it has MLKIt implement for barcode

LightningXCE, 2019-04-17:
Thank you both for your replies!

@sibelius, do I need to do anything special for that to work, or just call the camera as per the standard documentation?

sibelius, 2019-04-17:
check docs

LightningXCE, 2019-04-17:
The docs only specify that you can change "googleVisionBarcodeMode" to "inverted" only for Android. Would I want to explore https://github.com/react-native-community/react-native-camera/tree/master/examples/mlkit ?

Thank you again for your help