react-native-camera/react-native-camera #1553: FaCC Feature breaks backwards compatibility

delta98, 2018-05-09:
### Which implementation are you using

*RNCamera*

Given an example of the child implementation before FaCC
```
<RNCamera><View><Text>Child View</Text></View></RNCamera>
```

### Expected behaviour
The latest version of React Native Camera should be backwards compatible 

### Actual behaviour

`renderChildren` returns null if `hasFaCC` isn't provided, breaking backwards compatibility which was the original intention on the issue and PR for the feature.

Instead of returning `null` I believe it would be better to return `this.props.children` if a function isn't provided.

delta98, 2018-05-09:
@AndreiCalazans PR here #1554

AndreiCalazans, 2018-05-09:
Awesome @delta98  thank you very much!

PR #1554 (2018-05-09): UPDATED renderChildren to return this.props.children instead of null
Fixes #1553
closed 2018-05-09 by AndreiCalazans