react-native-camera/react-native-camera #1555: Impossible to build with React Native Stripe ?

LucasLpr, 2018-05-09:
### Which implementation are you using

*RNCamera* 

### Environment
- **Npm Version**: 5.6.0
- **React Native version**: 0.55.4
- **Gradle 3**

### react-native-camera
**Version**: 1.1.2

### Issue

After installed tipsi-stripe and builded successfully I tried to install react-native-camera.
And then it was impossible to me to build, I tried many solutions but nothing to do. I'am wondering if there isn't any conflicts between Stripe and React-Native-camera due to dependencies ?

this is my error message : 
> > Task :app:transformDexArchiveWithExternalLibsDexMergerForDebug FAILED
> D8: Program type already present: com.google.android.gms.internal.zzdkv
> 
> 
> FAILURE: Build failed with an exception.
> 
> * What went wrong:
> Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForDebug'.

And these are my dependencies : 
```
dependencies {
    implementation fileTree(dir: "libs", include: ["*.jar"])
    implementation "com.facebook.react:react-native:+"
    implementation (project(':react-native-maps'))
    implementation (project(':react-native-camera'))
    implementation "com.android.support:support-v4:27.1.0"
    implementation "com.android.support:appcompat-v7:23.0.1"
    implementation ('com.google.android.gms:play-services-vision:12.0.1') {
        force = true
    }
    implementation ("com.google.android.gms:play-services-base:+") {
        force = true;
    }
    implementation ("com.google.android.gms:play-services-maps:10.0.1") {
        force = true;
    }
}
```

```
 compileSdkVersion 26
    buildToolsVersion "26.0.2"
    defaultConfig {
        applicationId "com.myapp"
        minSdkVersion 16
        targetSdkVersion 26
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
    }
```

sibelius, 2018-05-09:
you need to exclude `com.google.android.gms` from both libraries and `force` a specific version in your application code

closed 2018-05-09 by sibelius

LucasLpr, 2018-05-09:
same error with these dependencies : 

> dependencies {
>     implementation fileTree(dir: "libs", include: ["*.jar"])
>     implementation project(':tipsi-stripe')
>     implementation "com.facebook.react:react-native:+"
>     implementation (project(':react-native-maps'))
>     {
>         exclude group: 'com.google.android.gms', module: 'play-services-base'
>         exclude group: 'com.google.android.gms', module: 'play-services-maps'
>     }
>     implementation (project(':react-native-camera')) {
>         exclude group: "com.google.android.gms"
>         implementation 'com.android.support:exifinterface:27.1.0'
>         
>     }
>     implementation "com.android.support:support-v4:27.1.0"
>     implementation "com.android.support:appcompat-v7:23.0.1"
>     implementation ('com.google.android.gms:play-services-vision:12.0.1') {
>         force = true
>     }
>     implementation ("com.google.android.gms:play-services-base:+") {
>         force = true;
>     }
>     implementation ("com.google.android.gms:play-services-maps:10.0.1") {
>         force = true;
>     }
> }

LucasLpr, 2018-05-09:
In the official documentation they say for dependencies : 

```
compile (project(':react-native-camera')) {
    exclude group: "com.google.android.gms"
    compile 'com.android.support:exifinterface:25.+'
    compile ('com.google.android.gms:play-services-vision:12.0.1') {
        force = true
    }
}
```

You think that I have to exclude group : "exclude group: "com.google.android.gms" in implem of stripe also ?

sibelius, 2018-05-09:
try to debug using this approach:
https://medium.com/mindorks/avoiding-conflicts-in-android-gradle-dependencies-28e4200ca235

LucasLpr, 2018-05-09:
Thank you Sibelius, I followed your link. I make a new step.
The new error message is  : 

> Execution failed for task ':app:preDebugBuild'.
> > Android dependency 'com.android.support:support-v4' has different version for the compile (27.0.0) and runtime (27.1.0) classpath. You should manually set the same version via DependencyResolution

My question is : where do I know which one is using : com.android.support ?

sfyfe31, 2018-10-22:
@LucasLpr did you ever figure this out? I'm having the same issue. I can link `tipsi-stripe` and `react-native-camera` separately fine, but together I just can't get my app to run. I've tried excluding `com.google.android.gms` from both and then force declaring a version, but I just get `AppName has stopped` when it tries to run.

LucasLpr, 2018-10-23:
@sfyfe31 Yes, it has been a long time haha :3 However it took me lots of time to make my dependencies working together :