react-native-camera/react-native-camera #1556: Face detection has not been included in this build.

zdenham, 2018-05-10:
### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?
RNCamera

### Steps to reproduce
1. yarn add react-native-camera
2. add to podfile: `pod 'react-native-camera', subspecs: ['RCT', 'RN', 'FaceDetector'], path: '../node_modules/react-native-camera'`
3. pod install
4. Follow directions for installing GVM frameworks, restart xcode and build
5. call:
```
imgPath => {
  const options = {
    mode: FaceDetector.Constants.Mode.fast,
  };

  return FaceDetector.detectFacesAsync(imgPath, options);
};
```

### Does it work with Expo Camera?
RNCamera is based on Expo Camera, if this does not work with Expo Camera it won't work with RNCamera as well.
https://forums.expo.io/t/building-react-native-camera-with-facedetector/9671

### Expected behaviour
Tell us what should happen
Promise resolves with an object of face data.

### Actual behaviour
Tell us what happens instead
Promise resolves with message: "Face detection has not been included in this build."

### Environment
- **Node.js version**: 9.0.0
- **React Native version**: 0.55.3
- **React Native platform + platform version**: iOS 11.3

### react-native-camera
npm version

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

zdenham, 2018-05-10:
This is resolved in the master branch if you manually link!

closed 2018-05-10 by zdenham

Tayyaba3317, 2020-11-09:
@zdenham, I am facing this issue for version:

"react-native-camera": "3.40.0"
"react-native": "0.63.2"

I there anything i am missing?

Thanks