react-native-camera/react-native-camera #1557: `takePictureAsync` never returns

pstanton, 2018-05-10:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Gradle 3
This package required Gradle 3, follow this migration guide https://developer.android.com/studio/build/gradle-plugin-3-0-0-migration

### Which implementation are you using

*RNCamera*

**Steps to reproduce**

```
	onTakePhoto = async function() {
		console.log("Camera.onTakePhoto");
		if (!this.camera) {
			Alert.alert("Error", "No camera!");
			return;
		}
		const data = await this.camera.takePictureAsync({ quality: 0.8, base64: true, fixOrientation: true, width: this.maxSize });
		console.log("taken", data );
	};
```
```
<RNCamera
					ref={ref => {
						this.camera = ref;
					}}
					style={{
						flex: 1
					}}
					type={RNCamera.Constants.Type.back}
					flashMode={RNCamera.Constants.FlashMode.on}
					permissionDialogTitle={"Permission to use camera"}
					permissionDialogMessage={"We need your permission to use your camera phone"}
				/>
				<Button title="Take" onPress={this.onTakePhoto.bind(this)} />
```

When `onTakePhoto` is called by pressing the "take" button, the `await this.camera.takePictureAsync` never returns. I can see the flash come on do it appears the photo is taken, however I never receive the data.

There are no warnings or errors.

This code unchanged used to work before I upgraded react and react-native.

### Environment
- **Node.js version**:  node: '9.0.0',
- **React Native version**:
		"react": "^16.3.1",
		"react-native": "^0.55.4",
		"react-native-camera": "^1.1.2",
- **testing on Android Galaxy S6 SM-G920I - 7.0

### react-native-camera
**Version**: npm version 5.5.1

vink1987, 2018-06-06:
@pstanton I am facing the same issue. The method never returns anything. How did you overcome it?

pstanton, 2018-06-06:
@vink1987 sorry - a lot of this react native stuff is voodoo and i can't always figure out what fixes some of these issues!! i believe it would have been one of the following:

(windows commands - convert to your os)

```
cd android
gradlew clean
```
or 
```
rmdir /S node_modules
yarn
```

hope that helps!

vink1987, 2018-06-07:
@pstanton I was trying on Android 5 and 6. react-native-camera seems to be working fine on Android 7 or greater. My phone has Android 6. When I try to take picture, it acts weird and then I can't capture images even from the phone's camera application. Picture and Video capturing works fine on Android 7 and 8.

Which version are you using?