react-native-camera/react-native-camera #1561: Could not find method google() for arguments [] on repository container.

hungdt-ibl, 2018-05-16:
### Error
```
./gradlew assembleRelease
Incremental java compilation is an incubating feature.

FAILURE: Build failed with an exception.

* Where:
Build file '/Volumes/Data/az_com/node_modules/react-native-camera/android/build.gradle' line: 7

* What went wrong:
A problem occurred evaluating project ':react-native-camera'.
> Could not find method google() for arguments [] on repository container.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED
```

### Noted

Today i meet this error : after auto update from gradle

```
./gradlew assembleRelease
Download https://jcenter.bintray.com/com/android/tools/build/gradle/2.3.0/gradle-2.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/build/gradle-core/2.3.0/gradle-core-2.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/build/builder/2.3.0/builder-2.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/lint/lint/25.3.0/lint-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/build/gradle-api/2.3.0/gradle-api-2.3.0.pom
Download https://jcenter.bintray.com/com/android/databinding/compilerCommon/2.3.0/compilerCommon-2.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/build/builder-model/2.3.0/builder-model-2.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/build/builder-test-api/2.3.0/builder-test-api-2.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/sdklib/25.3.0/sdklib-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/sdk-common/25.3.0/sdk-common-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/common/25.3.0/common-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/build/manifest-merger/25.3.0/manifest-merger-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/ddms/ddmlib/25.3.0/ddmlib-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/analytics-library/protos/25.3.0/protos-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/analytics-library/shared/25.3.0/shared-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/analytics-library/tracker/25.3.0/tracker-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/lint/lint-checks/25.3.0/lint-checks-25.3.0.pom
Download https://jcenter.bintray.com/com/android/databinding/baseLibrary/2.3.0/baseLibrary-2.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/layoutlib/layoutlib-api/25.3.0/layoutlib-api-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/dvlib/25.3.0/dvlib-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/repository/25.3.0/repository-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/lint/lint-api/25.3.0/lint-api-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/annotations/25.3.0/annotations-25.3.0.pom
Download https://jcenter.bintray.com/com/android/tools/build/gradle/2.3.0/gradle-2.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/build/gradle-core/2.3.0/gradle-core-2.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/build/builder/2.3.0/builder-2.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/lint/lint/25.3.0/lint-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/build/gradle-api/2.3.0/gradle-api-2.3.0.jar
Download https://jcenter.bintray.com/com/android/databinding/compilerCommon/2.3.0/compilerCommon-2.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/build/builder-model/2.3.0/builder-model-2.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/build/builder-test-api/2.3.0/builder-test-api-2.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/sdklib/25.3.0/sdklib-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/sdk-common/25.3.0/sdk-common-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/common/25.3.0/common-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/build/manifest-merger/25.3.0/manifest-merger-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/ddms/ddmlib/25.3.0/ddmlib-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/analytics-library/protos/25.3.0/protos-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/analytics-library/shared/25.3.0/shared-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/analytics-library/tracker/25.3.0/tracker-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/lint/lint-checks/25.3.0/lint-checks-25.3.0.jar
Download https://jcenter.bintray.com/com/android/databinding/baseLibrary/2.3.0/baseLibrary-2.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/layoutlib/layoutlib-api/25.3.0/layoutlib-api-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/dvlib/25.3.0/dvlib-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/repository/25.3.0/repository-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/lint/lint-api/25.3.0/lint-api-25.3.0.jar
Download https://jcenter.bintray.com/com/android/tools/annotations/25.3.0/annotations-25.3.0.jar
Incremental java compilation is an incubating feature.
Download https://jitpack.io/com/github/PhilJay/MPAndroidChart/v3.0.3/MPAndroidChart-v3.0.3.pom
Download https://jitpack.io/com/github/PhilJay/MPAndroidChart/v3.0.3/MPAndroidChart-v3.0.3.jar

```

sibelius, 2018-05-16:
android/build.gradle

```gradle
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    repositories {
        jcenter()
        google()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.1.0'
        classpath 'com.google.gms:google-services:3.1.2'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven {
            url "https://maven.google.com" // Google's Maven repository
        }
    }
}
```

closed 2018-05-16 by sibelius

mjstelly, 2018-07-21:
Why was this issue closed? @sibelius It's still an issue. Your reply did nothing to resolve the problem. I'm confused.

n1ru4l, 2018-07-22:
@mjstelly updating to react-native v.0.56 should be enough to fix your configuration.