react-native-camera/react-native-camera #1562: Camera Crashes When Rendering

theebi, 2018-05-17:
Everytime when I render <Camera/> inside my view the app crashes with the error - JNI DETECTED ERROR IN APPLICATION: input is not valid Modified UTF-8: illegal start byte 0x98 from java.lang.String android.hardware.Camera.native_getParameters(). Any idea how can I overcome this?

jgfidelis, 2018-05-17:
Please provide more information like Android device and android version.

Do you have all permissions on your manifest?

theebi, 2018-05-17:
Yeah the camera permission is mentioned in the Manifest as well as Im request permission explicitly. The device is running android version 7.1.2 and its a Galaxy S5. I checked in other phone which runs 6.0.1 and in that it was working fine. Any idea why?

soninit, 2018-05-19:
I don't know why but my app crashed in all samsung devices too.
#1379 similar. Please check

closed 2019-06-08 by stale[bot]

Bardiamist, 2020-06-16:
My user V2 LTE (N6F26Q) (Android 7.1.1) got many crashes

```
android.hardware.Camera in native_getParameters
getParameters failed (empty parameters)
```