react-native-camera/react-native-camera #1564: App crashes when RNCamera gets rendered on iOS

TarikFojnica, 2018-05-17:
### Which implementation are you using

RNCamera

### Steps to reproduce

1. `npm install react-native-camera --save`
2. `react-native link react-native-camera`
3. Check in Xcode and make sure everything is linked properly. If not, follow the iOS Manual Install Guide.
4. Render it on the screen

### Expected behaviour
Render camera without crash.

### Actual behaviour

Once the `RNCamera` gets rendered on the screen, the app crashes without any error in the debug console.

This issue is happening on my other large app, but to check if it will work on a fresh new app, I initialised a new one with Native Code, and the issue is here again.

This issue started happening after I built the app. Before it worked fine. I spent several days installing and going back to older version of my app, but unsuccessfully.

I think the issue has to do something with my Xcode setup which got messed up after I built the app the first time. I already did several Xcode installations, tried with 9.3 and 9.4 beta.

### Environment
- **Node.js version**:  v9.11.1
- **React Native version**: 0.55.4
- **React Native platform + platform version**: iOS 10.3
- **Xcode: 9.3.1

### react-native-camera
**Version**: ^1.1.2

jgfidelis, 2018-05-17:
Can you provide more information like what is happening, is there any error information being provided by XCode or RN?

Is it iOS only?

Are you installing with FaceDetection or not?

TarikFojnica, 2018-05-17:
It was my bad. I overlooked the `Info.plist `changes because I didn't need them before since my iOS version was 9. Please close.

closed 2018-05-17 by sibelius

BrightonWeBill, 2018-11-16:
> It was my bad. I overlooked the `Info.plist `changes because I didn't need them before since my iOS version was 9. Please close.

what was causing the error

TarikFojnica, 2018-11-16:
Adding this to `Info.plist` fixed the thing.

```
<key>NSCameraUsageDescription</key>
<string>Your message to user when the camera is accessed for the first time</string>

<!-- Include this only if you are planning to use the camera roll -->
<key>NSPhotoLibraryUsageDescription</key>
<string>Your message to user when the photo library is accessed for the first time</string>

<!-- Include this only if you are planning to use the microphone for video recording -->
<key>NSMicrophoneUsageDescription</key>
<string>Your message to user when the microphone is accessed for the first time</string>
```