react-native-camera/react-native-camera #1565: duplicate entry: com/facebook/infer/annotation/Assertions.class

a289459798, 2018-05-17:
duplicate entry: com/facebook/infer/annotation/Assertions.class

sibelius, 2018-05-17:
can you provide more stacktrace? and your gradle configs?

ios-dev-newbie, 2018-05-18:
I'm facing the same issue @sibelius  did u find any solution?

jgcmarins, 2018-05-18:
we need a better stack trace.
can someone provide it, please?

jpudysz, 2018-05-18:
```
Uncaught translation error: java.lang.IllegalArgumentException: already added: Lcom/facebook/infer/annotation/Assertions;
Message{kind=ERROR, text=Uncaught translation error: java.lang.IllegalArgumentException: already added: Lcom/facebook/infer/annotation/Assertions;, sources=[Unknown source file]}
```

[image]

Maydar, 2018-05-18:
https://github.com/react-native-community/react-native-camera/commit/a4599093ea984e83f33bb92e33bb09e1078ddc01 - this commit causes build errors.

bbar, 2018-05-18:
[stacktrace.txt](https://github.com/react-native-community/react-native-camera/files/2018886/stacktrace.txt)

harrydema, 2018-05-19:
Same problem here

hwangjr, 2018-05-21:
I use the latest studio and build tool, same error here.
```
./gradlew aR --stacktrace --info

xxxxxxxxx

:app:transformClassesWithMultidexlistForRelease (Thread[Task worker for ':' Thread 4,5,main]) completed. Took 2.286 secs.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:transformClassesWithMultidexlistForRelease'.
> com.android.build.api.transform.TransformException: Error while generating the main dex list.

* Try:
Run with --debug option to get more log output. Run with --scan to get full insights.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':app:transformClassesWithMultidexlistForRelease'.
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:100)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:70)
        at org.gradle.api.internal.tasks.execution.OutputDirectoryCreatingTaskExecuter.execute(OutputDirectoryCreatingTaskExecuter.java:51)
        at org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter.execute(SkipUpToDateTaskExecuter.java:62)
        at org.gradle.api.internal.tasks.execution.ResolveTaskOutputCachingStateExecuter.execute(ResolveTaskOutputCachingStateExecuter.java:54)
        at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:60)
        at org.gradle.api.internal.tasks.execution.SkipEmptySourceFilesTaskExecuter.execute(SkipEmptySourceFilesTaskExecuter.java:97)
        at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:87)
        at org.gradle.api.internal.tasks.execution.ResolveTaskArtifactStateTaskExecuter.execute(ResolveTaskArtifactStateTaskExecuter.java:52)
        at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:52)
        at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:54)
        at org.gradle.api.internal.tasks.execution.ExecuteAtMostOnceTaskExecuter.execute(ExecuteAtMostOnceTaskExecuter.java:43)
        at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:34)
        at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker$1.run(DefaultTaskGraphExecuter.java:248)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:336)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:328)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:199)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:110)
        at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker.execute(DefaultTaskGraphExecuter.java:241)
        at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker.execute(DefaultTaskGraphExecuter.java:230)
        at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.processTask(DefaultTaskPlanExecutor.java:123)
        at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.access$200(DefaultTaskPlanExecutor.java:79)
        at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker$1.execute(DefaultTaskPlanExecutor.java:104)
        at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker$1.execute(DefaultTaskPlanExecutor.java:98)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionPlan.execute(DefaultTaskExecutionPlan.java:626)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionPlan.executeWithTask(DefaultTaskExecutionPlan.java:581)
        at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor$TaskExecutorWorker.run(DefaultTaskPlanExecutor.java:98)
        at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:63)
        at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:46)
        at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:55)
Caused by: java.lang.RuntimeException: com.android.build.api.transform.TransformException: Error while generating the main dex list.
        at com.android.builder.profile.Recorder$Block.handleException(Recorder.java:55)
        at com.android.builder.profile.ThreadRecorder.record(ThreadRecorder.java:104)
        at com.android.build.gradle.internal.pipeline.TransformTask.transform(TransformTask.java:212)
        at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:73)
        at org.gradle.api.internal.project.taskfactory.IncrementalTaskAction.doExecute(IncrementalTaskAction.java:46)
        at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:39)
        at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:26)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$1.run(ExecuteActionsTaskExecuter.java:121)
[... cut ...]
        at com.android.tools.r8.graph.LazyLoadedDexApplication$Builder.build(LazyLoadedDexApplication.java:121)
        at com.android.tools.r8.dex.ApplicationReader.read(ApplicationReader.java:122)
        at com.android.tools.r8.dex.ApplicationReader.read(ApplicationReader.java:86)
        at com.android.tools.r8.GenerateMainDexList.run(GenerateMainDexList.java:40)
        at com.android.tools.r8.GenerateMainDexList.run(GenerateMainDexList.java:110)
        at com.android.builder.multidex.D8MainDexList.generate(D8MainDexList.java:83)
        ... 45 more

* Get more help at https://help.gradle.org
```

hwangjr, 2018-05-21:
hope to fix this as soon as better.

sibelius, 2018-05-21:
Can somebody rollback that commit?

sibelius, 2018-05-21:
1.1.4

closed 2018-05-21 by sibelius

enieber, 2018-08-31:
I try to use the lib but I have the similar error

## react-native-camera

version: 1.2.0

## react-native

version: 0.56.0

## gradle 

version: 2.3.3

I can't update Gradle in my project to version > 3, this problem can affect in compile in the project?

HarshithaHJ, 2019-01-25:
Any Updates here facing same issue

n1ru4l, 2019-01-25:
@HarshithaHJ Please open a new issue with the issue template.

PR #1576 (2018-05-21): Revert "Fix runtime error missing facebook infer annotation"
Reverts react-native-community/react-native-camera#1559

fix #1565