react-native-camera/react-native-camera #1567: Runtime Problems with FaceDetection object

davidpricedev, 2018-05-17:
### Which implementation are you using

*RNCamera*

### Steps to reproduce

1. `yarn add react-native-camera`
1. `react-native link react-native-camera`
1. swap out the Podfile line with this: `pod 'react-native-camera', subspecs: ['RCT', 'RN', 'FaceDetector'], path: '../node_modules/react-native-camera'`
1. run `pod install`
1. download all 5 tar.gz google libraries using the links in the main react-native-camera readme
1. unzip all 5 to a "~/Downloads/gmv" folder (via `tar -zxvf ../<eachFile>.tar.gz`)
1. delete the "copy" and "BarcodeDetector" folders (and all the dummy.txt files)
1. In xcode right click on the Frameworks folder (I already have one) and add all 4 folders in the gmv folder (namely: Detector, FaceDetector, Frameworks, and MVDataOutput) with the copy-items and create-groups options. 
1. add the AddressBook framework in build phases -> link section
1. add the `-lz` option to the linker flags (I already have `-lc++` and `-Objc`options)
1. Cmd-Q to quit xcode and then restart it 
1. use xcode to run it on my phone
1. watch the app crash with "TypeError: cannot read property fast of undefined" triggered by `faceDetectionMode={RNCamera.Constants.FaceDetection.Mode.fast}`

### Does it work with Expo Camera?

N/A

### Expected behaviour

App should not crash and should work the same way the android side does.

### Actual behaviour

Red box error: "TypeError: cannot read property fast of undefined" triggered by `faceDetectionMode={RNCamera.Constants.FaceDetection.Mode.fast}`

### Environment

- **Node.js version**: v8.9.1
- **React Native version**: 0.53.3
- **React Native platform + platform version**: ios 11.2.6
- **Xcode version**: 9.3.1

### react-native-camera

**Version**: 1.1.2

### More details

It might be the same issue as [1494](https://github.com/react-native-community/react-native-camera/issues/1494)? 

Android works fine. Compiling for ios works fine. When I comment out everything that touches `RNCamera.Constants.FaceDetection` then the crash goes away in ios. I've followed the extra linking steps to the letter (as described above) 3+ times now with no change in outcome. I've also tried a few variations such as creating a new (second) Frameworks folder anyway. 

The resulting Frameworks folder looks like this: 
[image]

In the Build Phases -> Link section I'm seeing the following new things after following these steps: 

* AddressBook.framework
* GoogleMVDataOutput.framework
* GoogleUtilities.framework
* GoogleInterchangeUtilities.framework
* GoogleMobileVision.framework
* GoogleSymbolUtilities.framework
* FaceDetector.framework
* GoogleNetworkingUtilities.framework

In the linker flags section this is what I have: 

[image]

jgfidelis, 2018-05-17:
On iOS, can you access other RNCamera constants?

davidpricedev, 2018-05-17:
Yes, I can. At least these two: 

```javascript
                    type={RNCamera.Constants.Type.front}
                    flashMode={RNCamera.Constants.FlashMode.on}
```

jgfidelis, 2018-05-17:
If you may, post the result of `console.log(RNCamera.Constants)`

davidpricedev, 2018-05-17:
Here is what I got for the entire Constants object (on the emulator):

```json
{
  "Type": {
    "back": 1,
    "front": 2
  },
  "FlashMode": {
    "on": 1,
    "torch": 3,
    "off": 0,
    "auto": 2
  },
  "AutoFocus": {
    "on": 2,
    "off": 0
  },
  "WhiteBalance": {
    "incandescent": 5,
    "sunny": 1,
    "shadow": 4,
    "auto": 0,
    "cloudy": 2,
    "fluorescent": 6
  },
  "VideoQuality": {
    "720p": 2,
    "1080p": 1,
    "2160p": 0,
    "480p": 3,
    "4:3": 3
  },
  "VideoCodec": {
    "AppleProRes422": "apcn",
    "H264": "avc1",
    "HVEC": "hvc1",
    "JPEG": "jpeg",
    "AppleProRes4444": "ap4h"
  },
  "BarCodeType": {
    "itf14": "org.gs1.ITF14",
    "ean8": "org.gs1.EAN-8",
    "code93": "com.intermec.Code93",
    "pdf417": "org.iso.PDF417",
    "upc_e": "org.gs1.UPC-E",
    "code39": "org.iso.Code39",
    "code39mod43": "org.iso.Code39Mod43",
    "code128": "org.iso.Code128",
    "aztec": "org.iso.Aztec",
    "qr": "org.iso.QRCode",
    "interleaved2of5": "org.ansi.Interleaved2of5",
    "datamatrix": "org.iso.DataMatrix",
    "ean13": "org.gs1.EAN-13"
  },
  "FaceDetection": {}
}
```

davidpricedev, 2018-05-17:
I get exactly the same thing from the physical phone too.

davidpricedev, 2018-05-17:
I created a repro project here: https://github.com/davidpricedev/react-native-camera-issue-1567-repro

davidpricedev, 2018-05-17:
I tried again on that repro project without using pods, and it didn't make any difference.

davidpricedev, 2018-05-18:
I was able to get things working by entirely removing react-native-camera from my Podfile and following the manual ios installation instructions followed by the custom face detection linking.

It seems there is something about the podfile version of linking in react-native-camera that doesn't cooperate well with the google-mobile-vision libraries.

davidpricedev, 2018-05-18:
It looks like Google mobile vision is available as a [pod](https://developers.google.com/vision/ios/getting-started) itself. I tried the other way - ripping out all the custom linking and using pods exclusively. Things seem to mostly work. 

I ended up with version 1.3.0 of google mobile vision, instead of the 1.1.0 version that react-native-camera depends on. That might be the cause of the weirdness I'm seeing with face coordinates.

Here is my podfile now: 
```ruby
  pod 'GoogleMobileVision/FaceDetector'
  pod 'GoogleMobileVision/Detector'
  pod 'GoogleMobileVision/MVDataOutput'
  pod 'react-native-camera', subspecs: ['RCT', 'RN', 'FaceDetector'], path: '../node_modules/react-native-camera'
```
The only change I've made apart from that was manually adding the link to AddressBook.framework.

davidpricedev, 2018-05-18:
If I downgrade the version number to match the version of the downloads, then the coordinate issues go away.

Podfile: 
```ruby
  pod 'GoogleMobileVision/FaceDetector', '~> 1.1.0'
  pod 'GoogleMobileVision/Detector', '~> 1.1.0'
  pod 'GoogleMobileVision/MVDataOutput', '~> 1.1.0'
```

I'm thinking of putting together a PR to fix the documentation on the main readme to use this method when using pods.

closed 2018-05-18 by davidpricedev

sireesha-9, 2019-12-28:
I am also facing the same issue. In my case, I am not using any pod related to *GoogleMobileVision* and *onFacesDetected* is also not triggering. Any help here will be very much appreciated. Thank you.

FactorMouk, 2020-02-13:
Same problem here!