react-native-camera/react-native-camera #1570: taking photos in landscape mode is rotating the photo 90º (turning to portrait)

imspikey, 2018-05-18:
### Which implementation are you using
*RNCamera* 

### Steps to reproduce
1. Take a picture in landscape mode

3.

### Expected behaviour
Image should  look like its in landscape 
[image]

### Actual behaviour
image looks cropped and in portrait 
[image]

### Environment
- **Node.js version:9.7.1
- **React Native version:0.53.3
- **React Native platform + platform version**: iOS 11.3.1

### react-native-camera
**Version**: 1.1.2

jgfidelis, 2018-05-18:
can you try `forceUpOrientation: true` option?

imspikey, 2018-05-18:
I have tried with no success

JoseVf, 2018-05-18:
In my case it happens when rotation mode is enabled on phone's configuration, but when it's disabled image will keep it's orientation.

imspikey, 2018-05-18:
But rotation is disabled in my case. Also video recording dose not have this problem only image capture

aditya-rumbleon, 2018-05-21:
I have the same issue.

aditya-rumbleon, 2018-05-21:
forceUpOrientation: true does not work for me. Any suggestions?

almost, 2018-05-22:
I added an "orientation" option to takePhotoAsync, the PR just got merged. So you can pass in the orientation you want.

For me the problem was that the orientation lock was turned on on the phone which made it think it was always portrait even when a modal that forced to landscape (as I use for the photo view finder screen) was in view.

imspikey, 2018-05-23:
@almost   Thank you my issue has bean solved after setting orientation to portrait 👍

sibelius, 2018-05-23:
@imspikey can you send a PR to add this to the docs

mun3, 2018-06-04:
Can I ask how the `orientation` option should be used? I've tried setting it to `3`, but my `CameraManager.Orientation` is undefined. My `CameraManager` is `NativeModules.RNCameraModule`. Does another manager need to be used instead?

imspikey, 2018-06-04:
sibelius ok 👍

imspikey, 2018-06-04:
mun3 her is my code

takePicture = async function() {
        if (this.camera) {

            const options = {orientation:'portrait',forceUpOrientation: true, mirrorImage:(fix this part as appropriatef)this.type == RNCamera.Constants.Type.front ,  quality: 0.20, base64: true };
            const data = await this.camera.takePictureAsync(options)
            return data
        }
    };

mun3, 2018-06-04:
@imspikey Thanks

I believe that's the fix for iOS, but the same issue arises in Android.

almost, 2018-06-07:
> Can confirm it is still bugged on Android. Pretty disappointing.
> The old camera actually works fine, not sure why the regression occurred.

My patch adding the orientation prop was ios only. I intend to do the same
for android when i have time. Unless someone else beats me to it :)

I would also be good to fix the auto orientation to match the orientation
used by the preview.

anztrax, 2018-11-06:
in index.ts, there're  options that support the orientation :
```
  /** Android only */
    skipProcessing?: boolean;
    fixOrientation?: boolean;

    /** iOS only */
    forceUpOrientation?: boolean; 
```

you can use when using them for example (in android) : 
```
 const takeCameraOptions = {
        quality : 0.5,  //just in case want to reduce the quality too
        skipProcessing: false,
        fixOrientation: true
      };
      this.camera.takePictureAsync(takeCameraOptions).then(data => .... ).catch(err => ...)
```

meiqi1992, 2018-11-08:
> Can I ask how the `orientation` option should be used? I've tried setting it to `3`, but my `CameraManager.Orientation` is undefined. My `CameraManager` is `NativeModules.RNCameraModule`. Does another manager need to be used instead? I'm having this same problem on Android.

same problem! how to resovle?

sibelius, 2018-12-14:
recheck on 1.6.3

closed 2018-12-14 by sibelius

HauLuu, 2019-10-15:
I have the same issue. Any fix so far?

`      const options = {
        base64: true,
        fixOrientation: true,
        pauseAfterCapture: true,
      };

      const data = await this.camera.takePictureAsync(options);`

sergchernata, 2019-11-12:
Experiencing this issue on iOS. App and camera are both set to `landscapeLeft`. Works as it should on Android, but on iOS the image taken is flipped 180 degrees.

iOS works as it should only if set to the opposite - `landscapeRight`.

```
"react-native": "^0.60.3",
"react-native-camera": "^3.9.0",
```

amcm23, 2019-11-19:
> I have the same issue with Samsung Galaxy S9. Any fix so far?
> 
> ```
> const options = {
>         base64: true,
>         fixOrientation: true,
>         pauseAfterCapture: true,
>       };
> 
> const data = await this.camera.takePictureAsync(options);
> ```

I fixed it with ` const options = { quality: 1, base64: true, pauseAfterCapture: true, orientation:'auto',fixOrientation: true};`

fabriciosautner, 2020-06-01:
Thank you @sergchernata ,

Incredibly, the orientation problem in landscape mode is solved by inverting the "orientation" option in takePictureAsync.

That is, where landscapeRight is to place landscapeLeft and vice versa.