react-native-camera/react-native-camera #1575: Does the component have no control over the onBarCodeRead callback interval attribute?

giantss, 2018-05-21:
## status quo
When I use this component to scan the QR code, the `onBarCodeRead` callback frequency is very high, basically every one millisecond.
I try to use the following code to control the program not to repeat the second pass, but performed two or three times at the same time (milliseconds).
```
...
  constructor(props){
        super(props);
        this.isBarCodeRead = false; //here
}
...

    _barcodeReceived = (e) => {
        if(e.data){
            if(!this.isBarCodeRead){
                console.log('this.isBarCodeRead',this.isBarCodeRead);
                this.isBarCodeRead = true;
                //Need to execute code
                setTimeout(()=>{
                    this.isBarCodeRead = false;
                }, 5000);
            }

        }
    };

```
[image]

[image]
[image]
[image]

Isn't this a bug?
[#419](https://github.com/react-native-community/react-native-camera/issues/419)

bachlongkocanh, 2018-12-21:
`  i = 0;
    onBarCodeRead = (scanResult) => {
        this.i++
        if (this.i == 0) {
            if (scanResult.data != null) {
                this.setState({ ResultScan: scanResult.data })
                this.props.navigation.state.params.onSelect({ result: scanResult.data });
                this.props.navigation.goBack();
            }
        }
        return;
    }`
resolve for me.

n1ru4l, 2018-12-23:
We should definitely have some way to disable `onBarCodeRead` calls. This could block the react-router bridge and could result in bad performance

panda6412, 2020-06-17:
any resolution here ?

closed 2020-08-24 by stale[bot]