react-native-camera/react-native-camera #1581: stop failed.

arapocket, 2018-05-23:
Using RNCamera

### Steps to reproduce
1. Call recordAsync() on Android on a real device
2. ???
3. Get error

### Environment
- **Node.js version**: 8.9.4
- **React Native version**: 0.55.4
- **React Native platform + platform version**: Android 7.1.1

### react-native-camera
**Version**: npm 1.1.4

**I am not using any options with recordAsync()**

This may be of use
[https://stackoverflow.com/questions/16221866/mediarecorder-failed-when-i-stop-the-recording](https://stackoverflow.com/questions/16221866/mediarecorder-failed-when-i-stop-the-recording)

**In LogCat**
`05-23 19:42:01.710 6731-6731/? E/WifiCallingStatusContral: mccmnc?:  is card availabe?: false
05-23 19:42:01.710 6731-6731/? I/WifiCallingStatusContral: ltw--OperatorWifiCallEnabled=false--strReady=no
05-23 19:42:01.710 6731-6731/? D/WifiCallingStatusContral: not support wfc
05-23 19:42:01.720 630-14194/? E/audio_hw_primary: tfa9890_get_mode: tfa_mode=2 old_mode=2  i2s_clk_enable=1  retry=1
05-23 19:42:01.720 630-14194/? I/soundtrigger: audio_extn_sound_trigger_update_device_status: device 0x5f of type 1 for Event 0, with Raise=1
05-23 19:42:01.720 630-14194/? V/audio_hw_primary: stop_input_stream: exit: status(0)
05-23 19:42:01.720 630-14194/? V/audio_hw_primary: in_standby: exit:  status(0)
05-23 19:42:01.720 630-14194/? D/audio_hw_primary: in_set_parameters: enter: kvpairs=routing=0
05-23 19:42:01.720 630-14194/? V/audio_hw_primary: in_set_parameters: exit: status(0)
05-23 19:42:01.720 656-14193/? W/AudioSource: Spurious callback from AudioRecord. Drop the audio data.
05-23 19:42:01.720 656-14193/? W/AudioSource: Spurious callback from AudioRecord. Drop the audio data.
05-23 19:42:01.720 656-14193/? W/AudioSource: Spurious callback from AudioRecord. Drop the audio data.
05-23 19:42:01.720 656-14193/? W/AudioSource: Spurious callback from AudioRecord. Drop the audio data.
05-23 19:42:01.720 656-14193/? W/AudioSource: Spurious callback from AudioRecord. Drop the audio data.
05-23 19:42:01.720 656-14193/? W/AudioSource: Spurious callback from AudioRecord. Drop the audio data.
05-23 19:42:01.720 656-14221/? E/MPEG4Writer: The number of recorded samples is 0
05-23 19:42:01.720 656-14221/? W/MPEG4Writer: 0-duration samples found: 1
05-23 19:42:01.720 656-14221/? W/MPEG4Writer: 0-duration samples found: 1
05-23 19:42:01.720 656-14221/? I/MPEG4Writer: Received total/0-length (1/0) buffers and encoded 0 frames. - Audio
05-23 19:42:01.720 656-14221/? I/MPEG4Writer: Audio track drift time: 0 us
05-23 19:42:01.720 656-14180/? I/MediaCodecSource: encoder (audio) stopped
05-23 19:42:01.720 656-919/? D/MPEG4Writer: Audio track source stopped
05-23 19:42:01.720 656-919/? D/MPEG4Writer: Audio track stopped
05-23 19:42:01.720 656-919/? D/MPEG4Writer: Duration from tracks range is [0, 0] us
05-23 19:42:01.720 656-919/? D/MPEG4Writer: Stopping writer thread
05-23 19:42:01.720 656-14200/? D/MPEG4Writer: 0 chunks are written in the last batch
05-23 19:42:01.720 656-919/? D/MPEG4Writer: Writer thread stopped
05-23 19:42:01.720 656-919/? D/MPEG4Writer: Video track stopping
05-23 19:42:01.720 656-919/? D/MPEG4Writer: Audio track stopping
05-23 19:42:01.720 13736-13736/com.convoyerandroid E/MediaRecorder: stop failed: -1007
05-23 19:42:01.730 13736-13736/com.convoyerandroid E/unknown:ReactNative: Exception in native call
                                                                          java.lang.RuntimeException: stop failed.
                                                                              at android.media.MediaRecorder.stop(Native Method)
                                                                              at com.google.android.cameraview.Camera1.stop(Camera1.java:145)
                                                                              at com.google.android.cameraview.Camera1$1.onSurfaceDestroyed(Camera1.java:116)
                                                                              at com.google.android.cameraview.PreviewImpl.dispatchSurfaceDestroyed(PreviewImpl.java:60)
                                                                              at com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureDestroyed(TextureViewPreview.java:59)
                                                                              at android.view.TextureView.releaseSurfaceTexture(TextureView.java:249)
                                                                              at android.view.TextureView.onDetachedFromWindowInternal(TextureView.java:222)
                                                                              at android.view.View.dispatchDetachedFromWindow(View.java:15565)
                                                                              at android.view.ViewGroup.removeViewInternal(ViewGroup.java:4715)
                                                                              at android.view.ViewGroup.removeViewInternal(ViewGroup.java:4689)
                                                                              at android.view.ViewGroup.removeView(ViewGroup.java:4620)
                                                                              at org.reactnative.camera.RNCameraView.onViewAdded(RNCameraView.java:205)
                                                                              at android.view.ViewGroup.dispatchViewAdded(ViewGroup.java:4310)
                                                                              at android.view.ViewGroup.addViewInner(ViewGroup.java:4467)
                                                                              at android.view.ViewGroup.addView(ViewGroup.java:4258)
                                                                              at android.view.ViewGroup.addView(ViewGroup.java:4198)
                                                                              at com.facebook.react.uimanager.ViewGroupManager.addView(ViewGroupManager.java:45)
                                                                              at com.facebook.react.uimanager.NativeViewHierarchyManager.manageChildren(NativeViewHierarchyManager.java:400)
                                                                              at com.facebook.react.uimanager.UIViewOperationQueue$ManageChildrenOperation.execute(UIViewOperationQueue.java:179)
                                                                              at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:821)
                                                                              at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:928)
                                                                              at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:46)
                                                                              at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:988)
                                                                              at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:29)
                                                                              at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:134)
                                                                              at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:105)
                                                                              at android.view.Choreographer$CallbackRecord.run(Choreographer.java:872)
                                                                              at android.view.Choreographer.doCallbacks(Choreographer.java:686)
                                                                              at android.view.Choreographer.doFrame(Choreographer.java:618)
                                                                              at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:860)
                                                                              at android.os.Handler.handleCallback(Handler.java:751)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:95)
[... cut ...]
                                                                              at android.app.ActivityThread.main(ActivityThread.java:6152)
                                                                              at java.lang.reflect.Method.invoke(Native Method)
                                                                              at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
                                                                              at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
`

arapocket, 2018-05-24:
Turned out the issue was from [react-native-vector-icons](https://github.com/oblador/react-native-vector-icons)

I had an icon button for the record button that changed via a boolean setState. The icon would change from a camera to a stop icon. Somehow this re-rendering of the icon was forcing stopRecording() to be called. Very bizarre.

closed 2018-05-24 by arapocket

komaldeep, 2018-08-17:
I ma facing this issue .. how you solved in the end ??

watadarkstar, 2018-10-12:
@arapocket How did you solve this?

iCueto, 2018-11-07:
The component Camera can't get inside it another elements, if you have a layout like this:

```
<View>
  <Camera>
   <View>
 <Text>Element nested in Camera Preview</Text>
</View>
</Camera>
</View>
```

It will fail with the error: stop failed, that it's caused by the setState that re-render the element Camera and when the state change. Change it to something like this.

```
<View>
  <View>
     <Camera/>
     <View>
        <Text>Element below of Camera Preview</Text>
     </View>  
</View>
</View>
```