react-native-camera/react-native-camera #1584: Crash on iOS when trying to use face detection features

olegwn, 2018-05-25:
I'm using RNCamera.

### Steps to reproduce
1. Configure everything with cocoapods per https://github.com/react-native-community/react-native-camera guide, including GMV frameworks.
2) Include onFacesDetected callback in RNCamera component as a prop

### Does it work with Expo Camera?
I'm not using EXPO

### Expected behaviour
Face detection should work

### Actual behaviour
Red screen with error
[image]

and

the following errors in XCode logs:
```
[RNCamera updateFaceDetecting:]: unrecognized selector sent to instance 0x137d05800
[RCTUIManager.m:1095] Error setting property 'faceDetectorEnabled' of RNCamera with tag #102: Exception thrown while executing UI block: -[RNCamera updateFaceDetecting:]: unrecognized selector sent to instance 0x137d05800
[501:63056] Error setting property 'faceDetectorEnabled' of RNCamera with tag #102: Exception thrown while executing UI block: -[RNCamera updateFaceDetecting:]: unrecognized selector sent to instance 0x137d05800
```

### Environment
- **Node.js version**: v6.11.2
- **React Native version**: 0.55.4
- **React Native platform + platform version**: iOS 10 & iOS 11.3 (tried on two devices)

### react-native-camera
**Version**: 1.1.4

arapocket, 2018-06-14:
Yup got this as well.

sibelius, 2018-06-14:
manual installation works?

can u test again against master?

arapocket, 2018-06-14:
Manual installation of react-native-camera won't work for me. I get the error about too many header search paths.

Pod installation of the face detection fails. Whenever I run pod install I get 
target overrides the `FRAMEWORK_SEARCH_PATHS`

arapocket, 2018-06-14:
I got both react-native-camera and the face detection working with a pod install, but still got this error when opening the camera.

sibelius, 2018-06-14:
open another issue with this new issue

arapocket, 2018-06-14:
@sibelius I have the same exact issue as him. Did a pod install of everything and got the same message as him when opening the camera.

Is there another way to pass the onFacesDetected prop? Aren't we supposed to use an options object for all props in react-native-camera now?

nelson-10e, 2018-06-30:
Any updates on what to do here? I'm running into the same issue

arapocket, 2018-06-30:
@nelson-10e I wasn't able to get it working.

nelson-10e, 2018-07-01:
Thanks for the update @arapocket

basavarajvs, 2018-07-19:
Faced the same issue since 2 days but got it resolved..
Inqude. 
Below are my settings which helped me to build a **RNCamera app with FaceRecognition** enabled.

1. Did not use cocoapods.. Using cocoa pods crashed my app
2. Downloaded google vision files and followed instruction as mentioned in the link https://www.npmjs.com/package/react-native-camera , Readme. Without these files I was getting the Red screen as above. Without GoogleVision files the code tries to update faceDetection stuff which fails resulting in Red Screen as Above 
3. Versions used
    "react": "16.3.1",
    "react-native": "0.55.2",
    "react-native-camera": "^1.1.4"
4. react-native-camera 1.1.5 is unstable and recognition seems broken
5. Xcode 9.4.1

ayberkanilatsiz, 2018-07-25:
same issue

sospedra, 2018-10-03:
This fixed it for me:

```
pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
  'FaceDetector'
]
```

sibelius, 2018-10-03:
tks

closed 2018-10-03 by sibelius