react-native-camera/react-native-camera #1585: I got crash at onLayout in Android.

JJMoon, 2018-05-28:
I call getSupportedRatiosAsync for Android setup, which (I guess) result in crash as follows.

https://github.com/react-native-community/react-native-camera/blob/60b9ef1b84377f7102b0635b9e09b7c7b5511fe2/android/src/main/java/org/reactnative/camera/RNCameraView.java#L191

I get width / height ratio list and set camera aspect ratio to closest one. Some phone generates this error which I found Crashlytics. I guess this line needs some protective code.

sibelius, 2018-05-28:
please provide the fullstack trace from crashlytics

JJMoon, 2018-05-30:
`
com.facebook.react.bridge.ReactContext.handleException (ReactContext.java:311)
com.facebook.react.uimanager.GuardedFrameCallback.doFrame (GuardedFrameCallback.java:31)
com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame (ReactChoreographer.java:134)
com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame (ChoreographerCompat.java:105)
android.view.Choreographer$CallbackRecord.run (Choreographer.java:909)
android.view.Choreographer.doCallbacks (Choreographer.java:723)
android.view.Choreographer.doFrame (Choreographer.java:655)
android.view.Choreographer$FrameDisplayEventReceiver.run (Choreographer.java:897)
android.os.Handler.handleCallback (Handler.java:789)
android.os.Handler.dispatchMessage (Handler.java:98)
android.os.Looper.loop (Looper.java:164)
android.app.ActivityThread.main (ActivityThread.java:6938)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.Zygote$MethodAndArgsCaller.run (Zygote.java:327)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1374)
`

davidpricedev, 2018-06-01:
It doesn't happen on the (pixel 2 api 27) emulator though, only happens on the real 1+5t. It is triggered by the transition - i.e. when the other code is calling `activity.getWindow().getDecorView().setSystemUiVisibility()`

davidpricedev, 2018-06-01:
I switched to react-native-immersive which doesn't seem to cause this problem.

JJMoon, 2018-06-04:
Thanks @davidpricedev for your answer. But, I don't use r-n-fullscreen. My crashes happen on Samsung Galaxy 8 (and other devices)
One thing I mind is that I call `getSupportedRatiosAsync` to select adequate aspect ratio. I suspect the return lists of front / back camera's aspect ratios are not the same, which can cause the crashes. But, in my test phone the elements of the lists are the same. I guess it's not the case.
Any idea?

guguji5, 2018-06-24:
I have got the same error in Samsung Galaxy 5, doesn't anyone can help?

JJMoon, 2018-10-24:
The error happens in Huawei device, the message is.
`Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference`

getDouble
ReadableNativeArray.java 114

mmccartney, 2018-10-25:
Same here, running on emulator `Nexus_5X_API_27_x86` with code that doesn't even touch the result of the Promise.  See the usage below (obviously without the `~~~~` underlined 😄 )

```
<RNCamera
   style={styles.ScannerContainer}
   ref={(ref) => { this.camera = ref; ref.getSupportedRatiosAsync().then(() => {}); }}
      	       	      	      	      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   flashMode={RNCamera.Constants.FlashMode.on}
   captureAudio={false}
   ratio='1:1'
   permissionDialogTitle={'Permission to use camera'}
   permissionDialogMessage={'Gee Wish needs to use your camera to scan a barcode'}
   onBarCodeRead={this.onBarCodeRead}
/>                                                                                             
```

I get this stack trace:

```
Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
getDouble
    ReadableNativeArray.java:114
extractArgument
    JavaMethodWrapper.java:64
extractArgument
    JavaMethodWrapper.java:60
invoke
    JavaMethodWrapper.java:359
invoke
    JavaModuleWrapper.java:160
run
    NativeRunnable.java
handleCallback
    Handler.java:790
dispatchMessage
    Handler.java:99
dispatchMessage
    MessageQueueThreadHandler.java:29
loop
    Looper.java:164
run
    MessageQueueThreadImpl.java:192
run
    Thread.java:764
```

sibelius, 2019-04-01:
does this still happens?

BenjaminMidtvedt, 2019-05-05:
> does this still happens?

Still happening on Galaxy s9, on master branch and latest RN.

lata2694, 2019-07-10:
I got the same issue on samsung phone. Is there any update about it ?

JJMoon, 2019-07-16:
I updated module and react-native to 0.59. No more this crashes.
I guess, it's good to close this issue.

JJMoon, 2019-07-16:
I found weird action on Huawei tablet.
> I installed on the tablet.
> Go to the screen which shows camera permissions, but it doesn't show up. (error on Huawei)
> Use camera without any problem.
> Restart the app, and go the same screen, and app crashes (maybe there is no permission)

404-html, 2019-09-20:
Experienced this recently on following devices: 

P30 Pro (HWVOG)
Galaxy S10 (beyond1q)
Galaxy A8(2018) (jackpotlte)

Stack trace:
```java.lang.RuntimeException: 
  at com.google.android.cameraview.Camera1.setUpPreview (Camera1.java:201)
  at com.google.android.cameraview.Camera1$1.onSurfaceChanged (Camera1.java:131)
  at com.google.android.cameraview.PreviewImpl.dispatchSurfaceChanged (PreviewImpl.java:56)
  at com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureAvailable (TextureViewPreview.java:46)
  at android.view.TextureView.getTextureLayer (TextureView.java:398)
  at android.view.TextureView.draw (TextureView.java:344)
  at android.view.View.updateDisplayListIfDirty (View.java:19306)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at android.view.View.draw (View.java:20498)
  at android.view.View.updateDisplayListIfDirty (View.java:19306)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at com.facebook.react.views.view.ReactViewGroup.dispatchDraw (ReactViewGroup.java:684)
  at android.view.View.updateDisplayListIfDirty (View.java:19297)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at com.facebook.react.views.view.ReactViewGroup.dispatchDraw (ReactViewGroup.java:684)
  at android.view.View.updateDisplayListIfDirty (View.java:19297)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at com.facebook.react.views.view.ReactViewGroup.dispatchDraw (ReactViewGroup.java:684)
  at android.view.View.draw (View.java:20498)
  at android.view.View.updateDisplayListIfDirty (View.java:19306)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at com.facebook.react.ReactRootView.dispatchDraw (ReactRootView.java:225)
  at android.view.View.updateDisplayListIfDirty (View.java:19297)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at android.view.View.updateDisplayListIfDirty (View.java:19297)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at android.view.View.updateDisplayListIfDirty (View.java:19297)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at android.view.View.updateDisplayListIfDirty (View.java:19297)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at android.view.View.updateDisplayListIfDirty (View.java:19297)
  at android.view.View.draw (View.java:20223)
  at android.view.ViewGroup.drawChild (ViewGroup.java:4394)
  at android.view.ViewGroup.dispatchDraw (ViewGroup.java:4170)
  at android.view.View.draw (View.java:20498)
  at com.android.internal.policy.DecorView.draw (DecorView.java:970)
  at android.view.View.updateDisplayListIfDirty (View.java:19306)
  at android.view.ThreadedRenderer.updateViewTreeDisplayList (ThreadedRenderer.java:740)
  at android.view.ThreadedRenderer.updateRootDisplayList (ThreadedRenderer.java:746)
  at android.view.ThreadedRenderer.draw (ThreadedRenderer.java:861)
  at android.view.ViewRootImpl.draw (ViewRootImpl.java:3586)
  at android.view.ViewRootImpl.performDraw (ViewRootImpl.java:3382)
  at android.view.ViewRootImpl.performTraversals (ViewRootImpl.java:2714)
  at android.view.ViewRootImpl.doTraversal (ViewRootImpl.java:1636)
  at android.view.ViewRootImpl$TraversalRunnable.run (ViewRootImpl.java:7946)
  at android.view.Choreographer$CallbackRecord.run (Choreographer.java:1092)
  at android.view.Choreographer.doCallbacks (Choreographer.java:893)
  at android.view.Choreographer.doFrame (Choreographer.java:812)
  at android.view.Choreographer$FrameDisplayEventReceiver.run (Choreographer.java:1078)
  at android.os.Handler.handleCallback (Handler.java:907)
  at android.os.Handler.dispatchMessage (Handler.java:105)
  at android.os.Looper.loop (Looper.java:216)
  at android.app.ActivityThread.main (ActivityThread.java:7625)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:524)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:987)
Caused by: java.io.IOException: 
  at android.hardware.Camera.setPreviewTexture (Native Method)
  at com.google.android.cameraview.Camera1.setUpPreview (Camera1.java:198)
```

JJMoon, 2019-10-18:
I got crash report on Huawei P10 lite.

```
Fatal Exception: com.facebook.react.bridge.JSApplicationIllegalArgumentException: Error while updating property 'ratio' of a view managed by: RNCamera
       at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp(ViewManagersPropertyCache.java:95)
       at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty(ViewManagerPropertyUpdater.java:132)
       at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps(ViewManagerPropertyUpdater.java:51)
       at com.facebook.react.uimanager.ViewManager.updateProperties(ViewManager.java:37)
       at com.facebook.react.uimanager.NativeViewHierarchyManager.updateProperties(NativeViewHierarchyManager.java:136)
       at com.facebook.react.uimanager.UIViewOperationQueue$UpdatePropertiesOperation.execute(UIViewOperationQueue.java:95)
       at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:917)
       at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:1025)
       at com.facebook.react.uimanager.UIViewOperationQueue.access$2600(UIViewOperationQueue.java:46)
       at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:1085)
       at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:29)
       at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:166)
       at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:84)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:931)
       at android.view.Choreographer.doCallbacks(Choreographer.java:742)
       at android.view.Choreographer.doFrame(Choreographer.java:671)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:919)
       at android.os.Handler.handleCallback(Handler.java:761)
       at android.os.Handler.dispatchMessage(Handler.java:98)
       at android.os.Looper.loop(Looper.java:156)
       at android.app.ActivityThread.main(ActivityThread.java:6617)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:942)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:832)
```

Ashoat, 2019-11-04:
The Android implementation of several methods requires the JS side to pass in a "node handle" to the native side. This node handle is only set when the JS side receives the ref to the native side, but for some reason the native ref can sometimes only be set after the JS wrapper is set.

The `Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference` issue occurs when a method such as `getSupportedRatiosAsync`, `getCameraIdsAsync`, or `getAvailablePictureSizes` is called before the native ref is set.

closed 2020-01-11 by stale[bot]