react-native-camera/react-native-camera #1588: Project wont build. What should the gradle and project setup be with RNCamera?

timjam, 2018-05-28:
I just installed all the latest things: Android studio, react-native-cli, react-native-camera, system-images, gradles etc etc, but still I can't get react-native-camera work at all in any project. What is the preferred gradle and project configuration to get this package work?

### Steps to reproduce
0. npm install -g react-native-cli
1. react-native init testapp
2. npm install react-native-camera --save
3. yarn   **(without this react-native link wont work!)**
4. react-native link react-native-camera
5. change classpath to 'com.android.tools.build:gradle:3.0.0' in build.gradle
6. Change Gradle wrapper to 4.1 in gradle.properties  **(Required by the gradle 3.0.0)**
7. Removed buildToolsVersion from app/build.gradle  **(Supresses one warning)**

Now if you try to run react-native run-android you end up getting lots of errors. These I fixed with these solutions found from all over internet:

Added this snippet 

```
subprojects {
  project.configurations.all {
     resolutionStrategy.eachDependency { details ->
        if (details.requested.group == 'com.android.support'
              && !details.requested.name.contains('multidex') ) {
           details.useVersion "27.1.0"
        }
     }
  }
}

```
to root's build.gradle file to remove this error 

_"Android dependency 'com.android.support:support-v4' has different version for the compile (23.0.1) and runtime (27.1.0) classpath. You should manually set the same version via DependencyResolution"_

Then I added 
`android.enableAapt2=false`
to gradle.properties, because I was getting this error:

_Error: java.util.concurrent.ExecutionException: com.android.tools.aapt2.Aapt2Exception: AAPT2 error: check logs for details_

At this point I figured out that
a) setting aapt to false doesn't actually fix the problem, but hides the cause and I have no idea what's the actual cause
b) maybe I am doing something completely wrong right from the beginning?

### Does it work with Expo Camera?
I've managed to get Expo camera work. This problem is might not actually be a problem with the RNCamera itself, but rather it is not compatible with the latest gradle setups.

### Expected behaviour
By following the instructions given on the installation part, one should be able to use the package in their projects

### Actual behaviour
Project wont build

### Environment
Node 9.8.0
npm 5.6.0
react-native 0.55.4
Android 6, Android 8

### react-native-camera
react-native-camera 1.1.4

Dror-Bar, 2018-05-28:
THIS IS NOT A SOLUTION - But, this can be solved by using an older version of react-native-camera.

I was having the exact same issues as you did with a new react-native project, never ending compile errors with android studio such as those you mention.

Revert what you upgraded in Android Studio (I'm using classpath 'com.android.tools.build:gradle:2.2.3' for example), or start a new project.

Use an older version, such as "react-native-camera": "1.0.3" in package.json. This should compile properly for you.

obsidianart, 2018-05-28:
After a lot of time and getting help from a friend who knows a lot more than me about android:
app/build.gradle
```
android {
    compileSdkVersion 27

    // org.gradle.caching = true

    defaultConfig {
        applicationId "com.cryptoviewer"
        minSdkVersion 16
        targetSdkVersion 23
        versionCode 196
        versionName "16.83"
        // ndk {
        //     abiFilters "armeabi-v7a", "x86"
        // }
    }
```
and dependencies
```
dependencies {
    implementation project(':react-native-camera')
   //...
    implementation "com.android.support:appcompat-v7:26.1.0" // <= YOU ONLY WANT THIS
    implementation "com.facebook.react:react-native:+"  // From node_modules
}
```

in build.gradle
```
allprojects {
   //...
    configurations.all {
        resolutionStrategy.force "com.android.support:support-v4:26.1.0"
    }
```

in gradle.properties
```
android.useDeprecatedNdk=true
android.enableAapt2=false
org.gradle.jvmargs=-Xmx4608M
```

Dror-Bar, 2018-05-29:
@timjam 
Did you find a solution already? obsidianart might work for you. As for my suggestion:
Did you also do `react-native link react-native-camera` after installing the package?
It should take care of almost everything for you. But make sure that you have the following:

This line inside the dependencies block in `android/app/build.gradle`:
```
dependencies {
    compile project(':react-native-camera')
    ...
}
```
These lines to `android/build.gradle` (I don't think this is done automatically):
```
allprojects {
    repositories {
        ...
        maven { url "https://jitpack.io" }
        maven { url "https://maven.google.com" }
    }
}
```

obsidianart, 2018-05-29:
That's my buildscript section
```
buildscript {
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.1.2'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}
```
And yes, I'm using the latest gradle

timjam, 2018-05-29:
Oh, okay. I need to test that later today.

obsidianart, 2018-05-29:
let me know if I can help in any other way. The problem I had was a little different (probably because I target a lowest device) but I hope my configuration can help.

Dror-Bar, 2018-05-29:
@timjam 
I'm sorry it didn't work for you. Could you perhaps post your `android/app/build.gradle`?

jgfidelis, 2018-05-29:
Hey, what are the issues you are currently facing? About the `google()` problem:
https://github.com/react-native-community/react-native-svg/issues/584

timjam, 2018-05-29:
I haven't been able to test those other solutions yet, but I guess I already know what's the problem. The gradle folder in Windows is on default at C:\Users\<user name>\.gradle. I have some non-Ascii characters in my username and I can't change the user name from the user folder and gradle doesn't like non-ascii characters at all. So I need to figure out how I can change the gradle config to use some other folder than that default one.

JakeRawr, 2018-06-15:
with `android.enableAapt2=false` in gradle.properties

I'm getting 
```
WARNING: The option 'android.enableAapt2' is deprecated and should not be used anymore.
Use 'android.enableAapt2=true' to remove this warning.
It will be removed at the end of 2018.
```

timjam, 2018-06-18:
@JakeRawr The warning is okay and you should definitely use 'andoid.enableAapt2=true'. Setting it to false will just hide possible "symptoms". Aapt2 is going to be the base builder in the future. The Aapt2 Error comes all the way from gradle. You can probably find more about this here:

https://stackoverflow.com/questions/50572845/fix-aapt2-error-in-android-studio-with-non-ascii-characters-in-windows-user-name

mashwinprdxn, 2018-06-21:
In case anyone's facing this issue on Android, please try the [gradle upgrade guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/GradleUpgradeGuide.md). It worked for me.

sharibtanweer, 2018-10-31:
No any solution in internet working. why? in android studio 3.2.1

timjam, 2018-10-31:
@sharibtanweer What have you tried so far? Upgrading to latest gradle should fix the issue. What's your setup?