react-native-camera/react-native-camera #1591: [Android] When using this package with react-native-camera get {"framesToPop": 1, "code": EUNSPECIFIED}

watadarkstar, 2018-05-29:
I get this error on Android only when I try to upload a file using a `uri` from `react-native-camera`:

```
{"framesToPop": 1, "code": EUNSPECIFIED}
```

My upload code:
```
// Attempt to start file upload
  try {
    uploadId = await Upload.startUpload(options);
  } catch (e) {
    console.log("failing here");<--------------------------------
    dispatch(messageFailed(uploadingMessage, I18n.t("messageFailedUpload"), e));
  }
```

My camera code:
```
startRecording = () => {
    if (this.camera) {
      this.camera
        .recordAsync({})
        .then(data => {
          console.log("data", data);
          this.props.upload(data.uri);
          this.navigateToChat();
        })
        .catch(err => console.error(err));
      this.setState({
        isRecording: true
      });
    }
  };
```

Not sure if this is a problem in this package or with `react-native-background-upload`. Any ideas as to what this error message means?

A workaround I have for now is: I copy the file using `RNFS.copyFile(uri,dest)` and use the destination `uri` (dest) and feed that to the uploader. But it would be nice if I could get rid of this workaround and just use the `uri` provided by `react-native-camera`.

ByteCrunch, 2018-08-02:
I have the same problem now but watadarkstar's workaround doesn't fix it for me.
But I got it working inside my Camera component before.
Is there any problem with my code ?

services/backgroundUpload.js:

`import { Alert, Platform } from 'react-native';
import Upload from 'react-native-background-upload';
import deviceStorage from "./deviceStorage";
import {throttle} from "lodash";
const backend = require('../lib/api/backend');
import path from 'react-native-path'
var RNFS = require('react-native-fs');

const backgroundUpload = {

    startUpload(opts) {
        try {
            Upload.getFileInfo(opts.path).then((metadata) => {
                (async () => {
                    let token = '';
                    let loggedIn = await deviceStorage.getKey("loggedIn");

                    if (loggedIn === "true")
                        token = await deviceStorage.getKey("token");

                    const options = Object.assign({
                        method: 'POST',
                        headers: {
                            'content-type': metadata.mimeType,
                            'Authorization': 'bearer ' + token
                        }
                    }, opts)

                    Upload.startUpload(options).then((uploadId) => {
                        Upload.addListener('progress', uploadId, (data) => {
                            //this.handleProgress(+data.progress)
                            console.log(`Progress: ${data.progress}%`)
                        })
                        Upload.addListener('error', uploadId, (data) => {
                            console.log(`Error: ${data.error}%`)
                        })
                        Upload.addListener('completed', uploadId, (data) => {
                            console.log('Completed!')
                        })
                    })
                })();
            })
        } catch (e) {
            Alert.alert('Upload error', JSON.stringify(e));
        }
    },

    initUpload(fileToUpload, childrenIds, categoryId, description) {

        // Move file to temp location (so it's gone from the camera roll)
        let filename = path.basename(fileToUpload);
        let origin = (Platform.OS === 'android') ? fileToUpload.replace("file://", "") : fileToUpload;
        let dest = RNFS.TemporaryDirectoryPath + '/' + filename
        RNFS.moveFile(origin, dest);

        // Upload image
        const options = {
            url: backend.bUrl('/media/upload'),
            path: dest,
            type: 'multipart',
            field: 'uploaded_media',
            parameters: {
                children: childrenIds,
                category: categoryId,
                description: description,
                filename: filename
            },
            // Below are options only supported on Android
            notification: {
                enabled: (Platform.OS === 'android')
            }
        }
        this.startUpload(options);
    }

};

export default backgroundUpload;`

meiqi1992, 2018-09-20:
same question , anyway to resolve it?

GreenRidingHood, 2019-01-29:
is there any progress?? i get same problem after I change my SDK or update everything at Android,

closed 2019-06-08 by stale[bot]