react-native-camera/react-native-camera #1598: Mocking react-native-camera with Jest

Tsiniloiv, 2018-05-31:
### Which implementation are you using

Both!

I'm having a lot of trouble figuring out how to properly mock react-native-camera when testing my app, because I'm using both RCTCamera for Android, and RNCamera for iOS.

I've managed to mock RCTCamera like so:

```
import React from 'react';

const constants = constants = {
  Aspect: {},
  BarCodeType: {},
  Type: {back: "back", front: "front"},
  CaptureMode: {},
  CaptureTarget: {},
  CaptureQuality: {},
  Orientation: {},
  FlashMode: {},
  TorchMode: {}
};

export class Camera extends React.Component {

  static constants = constants
  render() {
    return null;
  }
}

Camera.constants = constants;

export default Camera;
```

And then set it with

`jest.mock('react-native-camera', () => require.requireActual('../../__mocks__/react-native-camera').default);`

But that still leaves RNCamera undefined, and I can't seem to find or figure out a way to make it work both ways.

Any solutions or advice on how to mock both components with Jest, and not just one? I am very new to Jest.

### Environment
- **Node.js version**: 9.3.0
- **React Native version**: 0.55.2
- **React Native platform + platform version**: Android API 23

### react-native-camera
**Version**: 1.1.1

ivanbtrujillo, 2018-06-13:
Hi @Tsiniloiv, this is how I mock react-native-camera. I'm using version 1.1.4 (latest):

Create your mock inside your __mock__ folder:

react-native-camera.js
```
import React from 'react'

export class RNCamera extends React.Component {
  static Constants = {
    Aspect: {},
    BarCodeType: {},
    Type: { back: 'back', front: 'front' },
    CaptureMode: {},
    CaptureTarget: {},
    CaptureQuality: {},
    Orientation: {},
    FlashMode: {},
    TorchMode: {},
  }
  render() {
    return null
  }
}

export default RNCamera

```

You don't need to do anything else in your test, because jest will use the mock in your test instead of the native module.

Here you have an example of my test:

```
import React from 'react'
import Adapter from 'enzyme-adapter-react-16'
import { shallow, configure } from 'enzyme'
import PhotoCamera from '../'

configure({ adapter: new Adapter() })

describe('PhotoCamera Tests', () => {
  test('renders correctly', () => {
    const wrapper = shallow(<PhotoCamera />)
    expect(wrapper).toMatchSnapshot()
  })
})

```
Here you have some snapshots about it:

If I comment for example, the Constant.Type on the mock, it doesn't work:
[image]

But if I un-comment that line, it works as expected:
[image]

In my component, I use RNCamera following the docs, I don't make any stranger thing:

```
import { RNCamera } from 'react-native-camera'
....
....
```

Hope it helps!

sibelius, 2018-06-13:
Can u send a Pr adding this to docs?

ivanbtrujillo, 2018-06-14:
Yep of course!

sibelius, 2018-06-14:
fixed here https://github.com/react-native-community/react-native-camera/blob/master/docs/tests.md

closed 2018-06-14 by sibelius