react-native-camera/react-native-camera #1599: Can't write duplicate zip entry classes.jar:com/google/android/cameraview/AspectRatio$1.class

watadarkstar, 2018-05-31:
I get this error when I run `./gradlew assembleRelease` and sometimes when I make the project in Android studio: 

`Can't write [/workspace/github/underwriter-bot-mobile/a│
ndroid/app/build/intermediates/transforms/proguard/dev19/release/0.jar] (Can't read [~/workspace/github/underwriter-bo│
t-mobile/node_modules/react-native-camera/android/build/intermediates/intermediate-jars/release/classes.jar(;;;;;;**.class)] (Duplicate │
zip entry [classes.jar:com/google/android/cameraview/AspectRatio$1.class]))`

May be relevant I recently upgraded from gradle 2 to 3 following the the upgrade gradle instructions in this repo.

See `build.gradle` file in comments.

### Which implementation are you using

*RNCamera* 

### Steps to reproduce
```
cd android
./gradlew assembleRelease
```

### Does it work with Expo Camera?
Yes.

### Expected behaviour
It builds.

### Actual behaviour
It errors out with `Duplicate │
zip entry [classes.jar:com/google/android/cameraview/AspectRatio$1.class`

### Environment
  OS: macOS High Sierra 10.13.3
  Node: 8.11.1
  Yarn: 1.5.1
  npm: 5.6.0
  Watchman: Not Found
  Xcode: Xcode 9.4 Build version 9F1027a
  Android Studio: 3.1 AI-173.4720617

Packages: (wanted => installed)
  expo: ^26.0.0 => 26.0.0
  react: 16.3.0-alpha.1 => 16.3.0-alpha.1
  react-native: https://github.com/expo/react-native/archive/sdk-26.0.0.tar.gz => 0.54.2

### react-native-camera
"react-native-camera": "^1.1.4",

watadarkstar, 2018-06-01:
I have a detached Expo project using ExpoKit. The expo camera is not cutting it for our users so we want to use `react-native-camera`. I have installed it for iOS and it works. I have got it working in a dev build from Android Studio. But when I try to build an APK using `./gradlew assembleRelease` it fails with:

```
Warning: Exception while processing task java.io.IOException: Can't write [/Users/adriancaroll/workspace/github/underwriter-bot-mobile/android/app/build/intermediates/transforms/proguard/dev19/release/0.jar] 
(Can't read [/Users/adriancaroll/workspace/github/underwriter-bot-mobile/node_modules/react-native-camera/android/build/intermediates/intermediate-jars/release/classes.jar(;;;;;;**.class)] 
(Duplicate zip entry [classes.jar:com/google/android/cameraview/AspectRatio$1.class]))
Thread(Tasks limiter_1): destruction
```

So, I tried to exclude `com.google.android.cameraview` but that didn't work in my `android/app/build.gradle`:

```
implementation (project(':react-native-camera')) {
      exclude group: "com.google.android.gms"
      exclude group: "com.google.android.cameraview"
  }
```

Here is my build.gradle dependencies:

```
dependencies {
  implementation (project(':react-native-camera')) {
      exclude group: "com.google.android.gms"
      exclude group: "com.google.android.cameraview"
  }
  implementation project(':instabug-reactnative')
  implementation project(':react-native-linear-gradient')
  implementation project(':react-native-i18n')
  implementation project(':react-native-background-upload')
  implementation project(':react-native-fs')
  implementation('io.branch.sdk.android:library:2.17.1@aar') {
        transitive = true;
    }
  implementation('io.fabric.sdk.android:fabric:1.4.2@aar') {
        transitive = true;
    }
  implementation('com.crashlytics.sdk.android:crashlytics:2.9.2@aar') {
      transitive = true;
    }
  implementation('com.crashlytics.sdk.android:answers:1.4.1@aar') {
        transitive = true;
    }
  implementation fileTree(dir: 'libs', include: ['*.jar'])

  implementation 'com.android.support:multidex:1.0.1'

  // Our dependencies
  // implementation 'com.android.support:appcompat-v7:26.0.1'

  // Our dependencies from ExpoView
  // DON'T ADD ANYTHING HERE THAT ISN'T IN EXPOVIEW. ONLY COPY THINGS FROM EXPOVIEW TO HERE.
  implementation 'com.android.support:appcompat-v7:26.0.1'
  implementation 'com.facebook.android:facebook-android-sdk:4.7.0'
  implementation('com.facebook.android:audience-network-sdk:4.22.1') {
    exclude module: 'play-services-ads'
  }
  provided 'org.glassfish:javax.annotation:3.1.1'
  implementation 'com.jakewharton:butterknife:7.0.1'
  implementation 'de.greenrobot:eventbus:2.4.0'
  implementation 'com.amplitude:android-sdk:2.9.2' // Be careful when upgrading! Upgrading might break experience scoping. Check with Jesse. See Analytics.resetAmplitudeDatabaseHelper
  implementation 'com.squareup.picasso:picasso:2.5.2'
  implementation 'com.google.android.gms:play-services-gcm:9.8.0'
  implementation 'com.google.android.gms:play-services-analytics:9.8.0'
  implementation 'com.google.android.gms:play-services-maps:9.8.0'
  implementation 'com.google.android.gms:play-services-auth:9.8.0'
  implementation 'com.google.android.gms:play-services-location:9.8.0'
  implementation 'com.google.android.gms:play-services-ads:9.8.0'
    annotationProcessor 'com.jakewharton:butterknife:7.0.1'
    annotationProcessor 'com.raizlabs.android:DBFlow-Compiler:2.2.1'
  implementation "com.raizlabs.android:DBFlow-Core:2.2.1"
  implementation "com.raizlabs.android:DBFlow:2.2.1"
  implementation "com.madgag.spongycastle:core:1.53.0.0"
  implementation "com.madgag.spongycastle:prov:1.53.0.0"
  debugCompile 'com.squareup.leakcanary:leakcanary-android-no-op:1.4-beta1'
  // debugCompile 'com.squareup.leakcanary:leakcanary-android:1.4-beta1'
  releaseCompile 'com.squareup.leakcanary:leakcanary-android-no-op:1.4-beta1'
  implementation 'com.facebook.device.yearclass:yearclass:1.0.1'
  implementation 'commons-io:commons-io:1.3.2'
  implementation 'me.leolin:ShortcutBadger:1.1.4@aar'
  implementation 'com.nostra13.universalimageloader:universal-image-loader:1.9.5'
  implementation 'com.theartofdev.edmodo:android-image-cropper:2.4.7'
  implementation 'com.yqritc:android-scalablevideoview:1.0.1'
  implementation 'commons-codec:commons-codec:1.10'
  implementation 'com.segment.analytics.android:analytics:4.3.0'
  implementation 'com.google.zxing:core:3.2.1'
  implementation 'net.openid:appauth:0.4.1'
  implementation 'com.airbnb.android:lottie:2.2.0'
  implementation 'io.branch.sdk.android:library:2.6.1'
  implementation('io.nlopez.smartlocation:library:3.2.11') {
    transitive = false
  }
  implementation 'com.android.support:exifinterface:26.0.1'
  implementation 'com.squareup.okhttp3:okhttp:3.4.1'
  implementation 'com.squareup.okhttp3:okhttp-urlconnection:3.4.1'
  implementation 'com.squareup.okhttp3:okhttp-ws:3.4.1'
  implementation 'com.squareup.okio:okio:1.9.0'

  // Testing
  androidTestCompile 'com.android.support.test.espresso:espresso-core:3.0.1'
  // We use a modified build of com.android.support.test:runner:1.0.1. Explanation in maven-test/README
  androidTestCompile 'com.android.support.test:runner:1.0.1'
  androidTestCompile 'com.android.support:support-annotations:26.0.1'
  androidTestCompile 'com.google.code.findbugs:jsr305:3.0.0'
  androidTestCompile 'com.android.support.test.uiautomator:uiautomator-v18:2.1.1'
  androidTestCompile 'com.azimolabs.conditionwatcher:conditionwatcher:0.2'

  implementation('host.exp.exponent:expoview:26.0.0@aar') {
    transitive = true
  }
}
```

When I run `./gradlew app:dependencies > dep.txt` I get the following file: https://drive.google.com/open?id=17NVabzxchS5uS9gk4tGPZujXH0eVxjMi

fenderslave, 2018-07-20:
@watadarkstar do have found a solution to this problem?

alexlevy0, 2018-07-20:
It's works for me ! I remove this line `compile 'com.github.react-native-community:cameraview:cc47bb28ed2fc54a8c56a4ce9ce53edd1f0af3a5'` from my app/build.gradle file. I noticed she wasn't here : `react-native-camera/blob/master/docs/GradleUpgradeGuide.md` even its there `react-native-camera/blob/master/README.md`

irtemed88, 2018-11-05:
Hey folks -- I had this same error and fixed it by adding an exclusion to Expoview:
```
  implementation('host.exp.exponent:expoview:30.0.0@aar') {
    transitive = true
    exclude group: 'com.squareup.okhttp3', module: 'okhttp'
    exclude group: 'com.squareup.okhttp3', module: 'okhttp-urlconnection'
    exclude group: 'com.google.android', module: 'cameraview'
  }
```

I _think_ the issue was cameraview being used by expokit, but also by react-native-camera. For my particular use case I was only using the camera implementation in react-native-camera so there wasn't any problem excluding it here.

closed 2019-06-08 by stale[bot]