react-native-camera/react-native-camera #1601: [Android] - onBarCodeRead

bdouram, 2018-05-31:
Hi! 
The RN-Camera good works when **onBarCodeRead**  isn't set.

**The speed of camera is good when:**
`<RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            style = {styles.preview}
            type={RNCamera.Constants.Type.back}
            flashMode={RNCamera.Constants.FlashMode.off}
            permissionDialogTitle={'Permission to use camera'}
            permissionDialogMessage={'We need your permission to use your camera phone'}
        />
`

**But this make my Android Camera slow frame:**

`<RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            style = {styles.preview}
            type={RNCamera.Constants.Type.back}
            flashMode={RNCamera.Constants.FlashMode.off}
            permissionDialogTitle={'Permission to use camera'}
            permissionDialogMessage={'We need your permission to use your camera phone'}
            onBarCodeRead = {(e)=> alert(e.data)} // my camera slowing image now.....
        />`

**My code have this configurations:**

Android 6.1

`"dependencies": {
    "react": "16.3.1",
    "react-native": "0.55.4",
    "react-native-camera": "^1.1.4"
  }`

Could you help me?
Thanks.

bdouram, 2018-05-31:
I think if this issue is because of the scan area: It is the total screen size of my smartphone. Am i correct?

If yes:
How can I continue with the full size of my smartphone screen and limit only the scan area?

zhenizhui, 2018-06-01:
@bdouram i think you can try this repository https://github.com/shifeng1993/react-native-qr-scanner.

It may help you although it still has some bug that I found.

bdouram, 2018-06-03:
@zhenizhui you solve my problem!

Thanks!

bdouram, 2018-06-03:
Closing the issue.

closed 2018-06-03 by bdouram