react-native-camera/react-native-camera #1603: [RCTCamera] Making the preview aspect ratio match the real picture aspect ratio

Tsiniloiv, 2018-06-01:
Hey y'all, I'm running into issues with phones that take pictures in 4:3 ratio. The preview fills the whole screen and appears accurate, but the final photo will come out looking cropped, because they're two different aspect ratios.

So far I've tried hammering in a method to expose the Camera API's getSupportedPictureSizes() in order to then calculate the ratio from there, but acquiring the camera instance has been and continues to be very unreliable for some reason.

Is there currently any way to match the preview and picture ratios? If not, can anyone point me to the right place to hack one in?

eqyiel, 2018-06-12:
@Tsiniloiv I had to do something like this too.  You could use this as a starting point: https://github.com/mango-chutney/react-native-camera/commit/40cb984ca292a536cdf575911a4ce823911bb7c9

eqyiel, 2018-06-15:
I opened a PR: https://github.com/react-native-community/react-native-camera/pull/1636

AndrewHenderson, 2018-10-01:
@Tsiniloiv I find that on Android the image has extra info on the right and left which makes it look like it was cropped, but actually has more information, not less. Hopefully, @eqyiel's PR can get merged soon.

Udbhav12, 2018-10-12:
@eqyiel - When will be your PR merged in master.I am using react-native-camera:1.3.0 and facing this aspect ratio problem in android

My options are { quality: 1, base64: true, fixOrientation: true, skipProcessing: true }

closed 2019-04-18 by sibelius