react-native-camera/react-native-camera #1604: Is it possible to access the current recording time?

drd0013, 2018-06-01:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our migration guide here https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

### Gradle 3
This package required Gradle 3, follow this migration guide https://developer.android.com/studio/build/gradle-plugin-3-0-0-migration

### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. start a video recording with recordAsync
2. Attempt to display the current timing of the video
3.

### Expected behaviour
Ideally, we'd like to be able to pass in a timer prop to RNCamera that it would update while recording a video to match the current recording time.

### Actual behaviour
In absence of this feature, we will either need to not display the recording time or using a timer directly in our component that could be off and display an unreliable recording time.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

Is there another option? Is this an enhancement that RNCamera would be open to?

jgfidelis, 2018-06-01:
I think this would be a nice feature to have. I don't know the best way of doing this but a reasonable way I can see is starting a recorder timer like in the update part of this comment:
https://stackoverflow.com/questions/13872973/how-can-i-get-time-of-my-recorded-audio-in-iphone

I don't know if the AVCapture classes have a nicer way (and correct) to do this.

For android, a similar approach could be made.

Then on the JS side you'd just implement a method to request the value of that varible but keep in mind that having the exact time would be difficult because all methods that are JS -> native are async, needing promises or callbacks to get the value from the native side.

drd0013, 2018-06-04:
so the implementation would probably need to be passing in a callback to the native code that gets triggered when the counter updates or something along those lines?
I'm not very familiar with the native code that backs react native though if no one else is interested in implementing this, I could give it a shot.

sibelius, 2018-06-04:
I think  it is better to implement this as an event instead

callback will be much slower

gvsakhil, 2020-06-18:
Do we have any solution for this in 2020?

renekorss, 2020-06-18:
@gvsakhil Alternatively you could create it yourself. Not sure how precise it is but here is example:
https://github.com/phuochau/react-native-beautiful-video-recorder/blob/master/lib/index.js
You can view screenshot on [README](https://github.com/phuochau/react-native-beautiful-video-recorder).

closed 2020-08-24 by stale[bot]