react-native-camera/react-native-camera #1607: Preview image rotated by 90 degrees after 4 rotations

mun3, 2018-06-04:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

RNCamera

### Steps to reproduce
1. Rotate clockwise (or counter clockwise) 4 times in Android (API 25 & 26) emulator

### Expected behaviour
Preview image restores to the original orientation.

### Actual behaviour
Preview image is rotated by 90 degrees:

Original orientation:
[image]

Final orientation (after 4 rotations):
[image]

### Environment
- **React Native version**: 0.51.0
- **React Native platform + platform version**: Android 7.1.1, 8.0

### react-native-camera
**Version**: "master"

6pm, 2018-06-11:
It is well known behavior for nexus 5x. try to add `fixOrientation: true`.

sibelius, 2018-12-14:
recheck on 1.6.3

closed 2018-12-14 by sibelius