react-native-camera/react-native-camera #1609: EXIF orientation always 0

mun3, 2018-06-05:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Take phone with phone in portrait
2. Take photo with phone in landscape

### Does it work with Expo Camera?
Unable to migrate to expo.

### Expected behaviour
Different EXIF orientation values.

### Actual behaviour
EXIF orientation is always 0.

### Environment
- **React Native version**: 0.51.0
- **React Native platform + platform version**: Android 7.x, 8.x

### react-native-camera
**Version**: "master"

StarkMike, 2018-06-06:
Using recordAsync to get a video - I get EXIF rotate: 90 all the time - similar issue, but different value

## Using RNCamera
React Native: 0.51.0
RN Camera Version: 1.1.4
Testing platform: ios 11.3.1 on iphone7

This causes landscape videos to not get displayed properly :(

SimonErm, 2018-09-26:
Normally exif orientation is 0, when the manufacturer don't use exif orientation and directly rotates the image, so it already has the right rotation.

SimonErm, 2018-09-26:
@StarkMike your problem on ios should be fixed with latest version

vvusts, 2018-11-14:
We still have this issue. We use master version RNCamera.
```
const options = { fixOrientation: true, base64: false, forceUpOrientation: true, quality: 0.9 };
const data = await this.camera.takePictureAsync(options);
```
**On android** we take photo in landscape mode, EXIF orientation is 0 and image preview is like it is taken in portrait mode. 
**If I take photo on IOS** in landscape and rotate phone to portrait image is rotated normally like on phone default camera.

Is there solution for this?

lucas-dolsan, 2018-12-03:
Also having this issue.

**Environment**: 
```
  react-native-cli: 2.0.1
  react-native: 0.53.3
  npm: 6.4.1
  react-native-camera: master
  android version: 8.0.1
  device: Motorola Moto(G5s)
/android/app/src/main/AndroidManifest.xml 
     ...
     android:configChanges="orientation"
     android:screenOrientation="portrait"
     ...
```

sibelius, 2018-12-14:
recheck on 1.6.3

closed 2018-12-14 by sibelius