react-native-camera/react-native-camera #1612: BUILD FAILED - error: collection element of type 'NSInteger'

Grandmac, 2018-06-06:
### Which implementation are you using

react-native-camera@1.1.4

### Environment
- **React Native version**: 0.55
- **React Native platform + platform version**: iOS 11.3

### Bug under iOS

When running `react-native run-ios`

`node_modules/react-native-camera/ios/RN/RNCameraManager.m:100:22: error: collection element of type 'NSInteger' (aka 'long') is not an Objective-C object
             @"off": AVCaptureVideoStabilizationModeOff,
                     ^
1 error generated.

** BUILD FAILED **`

carsonwu, 2018-06-06:
same problem here!

carsonwu, 2018-06-06:
Well, @Grandmac , I just find the code where caused the error and turned them all into NSNumber type and finally can successfully build the project. I'm not sure if this is a correct solution but i've tried the camera function in my app and seems it works fine.

cody-y, 2018-06-06:
Also experiencing this error. in RNCameraManager.m I had to change validVideoStabilizationModes to 
```   
+ (NSDictionary *)validVideoStabilizationModes
{
    return @{
             @"off": [NSNumber numberWithInt: AVCaptureVideoStabilizationModeOff],
             @"standard": [NSNumber numberWithInt: AVCaptureVideoStabilizationModeStandard],
             @"cinematic": [NSNumber numberWithInt: AVCaptureVideoStabilizationModeCinematic],
             @"auto": [NSNumber numberWithInt: AVCaptureVideoStabilizationModeAuto]
             };
}

sibelius, 2018-06-06:
@n1ru4l can  u check this?

sibelius, 2018-06-06:
you can try this example https://github.com/expo/camerja to test expo camera

we could also have the same example using react-native-camera to show the difference

`camerja-rncamera`

n1ru4l, 2018-06-06:
@sibelius I think we should keep this issue open until the problem is solved. While trying to fix (and test the feature). #1613

@Grandmac @carsonwu Is my assumption about you guys using the master branch right? These changes should not be part of `react-native-camera@1.1.4`.

sibelius, 2018-06-06:
This comment was to another issue, sorry about that

Closed the wrong one

n1ru4l, 2018-06-06:
@sibelius Yeah that answer seemed perfect to one I previously commented 😂

closed 2018-06-06 by jgfidelis