react-native-camera/react-native-camera #1613: RNCamera: Front Camera shows black screen in example app on iOS

n1ru4l, 2018-06-06:
While ttrying testing (and fixing the feature #1606) with the rn-camera-example application ( https://github.com/react-native-community/rncamera-example). I discovered the following problems:

**Steps to reproduce:**

1. Clone `rn-camera-example` at the current master (https://github.com/react-native-community/rncamera-example/commit/4f87d52fe41be0bb85eb57b7e7f1456e2f7cdc5f)

```bash
git clone https://github.com/react-native-community/rncamera-example.git
cd rncamera-example
# checkout commit in which this occures (master at time of the posting)
git checkout 4f87d52fe41be0bb85eb57b7e7f1456e2f7cdc5f
```

2. Install `react-native-camera` at the version before #1606 got merged

```bash
 # install react-native-camera @
yarn add react-native-camera@https://github.com/react-native-community/react-native-camera.git#9cddb2a7213697663f6e53ad8c9603cd11c16754
```

3. Open xcode project and run App on device.

4. Use the toggle button to switch between cameras.

**Observation:**

- **The Front Camera only shows a black screen (sometimes the app crashes while switching)**
- **After the RNCamera Component shows a black screen pressing the SNAP button results in a application crash**

**Investigation:**

When I install the following version of react-native-camera (4a9322cb8b7d455fc28f7e67a15bff2fd9d7ea3e) aka the commit before the preview feature was implemented:

```bash
yarn add react-native-camera@https://github.com/react-native-community/react-native-camera.git#4a9322cb8b7d455fc28f7e67a15bff2fd9d7ea3e
```

The app crashes upon start, but after applying the following patch everything seems to be fine:

```diff
diff --git a/App.js b/App.js
index 47b560d..8440591 100644
--- a/App.js
+++ b/App.js
@@ -192,9 +192,9 @@ export default class CameraScreen extends React.Component {
         zoom={this.state.zoom}
         whiteBalance={this.state.whiteBalance}
         ratio={this.state.ratio}
-        faceDetectionLandmarks={RNCamera.Constants.FaceDetection.Landmarks.all}
-        onFacesDetected={this.onFacesDetected}
-        onFaceDetectionError={this.onFaceDetectionError}
+        // faceDetectionLandmarks={RNCamera.Constants.FaceDetection.Landmarks.all}
+        // onFacesDetected={this.onFacesDetected}
+        // onFaceDetectionError={this.onFaceDetectionError}
         focusDepth={this.state.depth}
         permissionDialogTitle={'Permission to use camera'}
         permissionDialogMessage={'We need your permission to use your camera phone'}
```

It seems like there a two different problems:

1. **The preview feature somehow breaked using the front camera**
2. **Using the face detection results in a bad access crash**

jgfidelis, 2018-06-06:
About the face detection: I investigated it a bit now. The crash seems to be at the google lib, so it is harder because it does not tell us the line that it happened. And it seems random... Sometimes it did not happen and detected faces normally.

jgfidelis, 2018-06-06:
About the front camera problem: I believe the problem is related to the pictureSize prop. The default pictureSize I used was `1920x1080` and I think this makes `[self.session canAddInput:input]` return NO and then the session is not updated. I will try to fix this tomorrow.

sibelius, 2018-06-07:
Can we upgrade google mobile vision sei as well?

n1ru4l, 2018-06-07:
Nice after these things are fixed I will update the app to include video recording features

jgfidelis, 2018-06-07:
Weird thing I am noticing.
On RNCamera-Example I changed the App.js to:
```
<RNCamera
        ref={ref => {
          this.camera = ref;
        }}
        style={{
          flex: 1,
        }}
        type={this.state.type}
        flashMode={this.state.flash}
        autoFocus={this.state.autoFocus}
        zoom={this.state.zoom}
        whiteBalance={this.state.whiteBalance}
        ratio={this.state.ratio}
        // faceDetectionLandmarks={RNCamera.Constants.FaceDetection.Landmarks.all}
        // onFacesDetected={this.onFacesDetected}
        onBarCodeRead={this.onBarCodeRead}
        // onFaceDetectionError={this.onFaceDetectionError}
        focusDepth={this.state.depth}
        permissionDialogTitle={'Permission to use camera'}
        permissionDialogMessage={'We need your permission to use your camera phone'}
        pictureSize={'1920x1080'}
      >
```

So I am setting pictureSize the same value as the defaultProp. And the front camera now works as expected. Weird.

jgfidelis, 2018-06-07:
I think #1618 fixes the camera flipping problems. When passing `None` as default, the camera behaves just like before.

About face detection, still did not try to fix. We need to find the last stable commit and check what changed.

jgfidelis, 2018-06-08:
Reopened because face detection still is not stable.

n1ru4l, 2018-06-13:
@jgfidelis are you planning to identify the issue?

sibelius, 2018-06-14:
@n1ru4l feel free to investigate as well

olegwn, 2018-06-28:
Yep, face detection will crash app at v1.1.5 of react-native-camera with this message and nothing else:

`2018-06-28 11:47:12.168172+0300 AppName[2569:586511] Instantiated <GMVFaceDetector: 0x17029f450>`

n1ru4l, 2018-06-29:
@sibelius I currently have no time to investigate further, our app is at the moment not relying on this feature, would be beneficial if a contributor that actually uses the face recognition could investigate.

davidpricedev, 2018-07-23:
I'm seeing something that sounds very similar to this. As suggested by the initial opening of this issue, things were working fine for me at version 1.1.2 of react-native-camera (which was before the 1606 PR).

I'm seeing the same thing in my log output window as @olegwn. 
In addition - when running in xcode I get the error below. 

Error in xcode in the assembly trace:
> Thread 32: EXC_BAD_ACCESS (code=1, address=0x116bd8000)

One difference from some of the other reports here is that it crashes the app every single time without fail. I'm using facial detection with the front camera which might be relevant.

```html
                <RNCamera
                    type={RNCamera.Constants.Type.front}
                    flashMode={RNCamera.Constants.FlashMode.on}
                    faceDetectionLandmarks={RNCamera.Constants.FaceDetection.Landmarks.all }
                    ...
                />
```

I created and applied a patch that rolls back all the ios-side to the state things were at as of 1.1.2. That resolves the issue. It sounds like maybe I could be more targeted and only hit the changes since 1.1.4 or maybe the particular commit in question here. 

At what point should we roll back this commit since it seems to be causing crashes and the feature it added presumably isn't yet widely used?

sibelius, 2018-07-23:
@davidpricedev nice job on debugging

can u send a PR to fix this? we can track and try to add the missing parts later on

jgfidelis, 2018-07-23:
A nice investigation would be: check from what version it started happening.

We now know that 1.1.2 is OK. See if 1.1.3 is also with a problem. Having the exact version it started happening, it would be easier to look into the commits that entered on that version and may have caused problems

jgfidelis, 2018-07-23:
1.1.3 is basically the same as 1.1.4, so it probably is one of this commits:
Merges between 1.1.2 and 1.1.3:
https://github.com/react-native-community/react-native-camera/commit/551a9d8d9356fdb7626bb54bd760689a0c7ab3b9
https://github.com/react-native-community/react-native-camera/commit/19697e96936d7a3306e9c4f6170d1732179569a1
https://github.com/react-native-community/react-native-camera/commit/7653092892465f4868a90dbef930071f65efb00f
https://github.com/react-native-community/react-native-camera/commit/68106f4e2c65d0047a19c2c4ef48fd2480907471
https://github.com/react-native-community/react-native-camera/commit/560c80124bd609c7d7357c3bac84769a93216aa9
https://github.com/react-native-community/react-native-camera/commit/86125a0a79f60c6774852772262ded3398a5c73b
https://github.com/react-native-community/react-native-camera/commit/0d889b9919b891d7cb32fcdfb305fae80a66bfa5
https://github.com/react-native-community/react-native-camera/commit/a4599093ea984e83f33bb92e33bb09e1078ddc01
https://github.com/react-native-community/react-native-camera/commit/ef362a2d577da286840f7521d128433038fae554
https://github.com/react-native-community/react-native-camera/commit/be8d8cc26c4ef0182b8358af209c7bdba2f2fae5
https://github.com/react-native-community/react-native-camera/commit/a2aee1eb98ace5261c19e490f569c3cdaa5d8e70
https://github.com/react-native-community/react-native-camera/commit/b84436d3ad46350f76e0a05d9a6438967a71dbeb

You can narrow down even more by looking only at commits that modify ios files.

davidpricedev, 2018-07-23:
I tried a patch for just the 1606 PR, but that doesn't fix things for me.
[PR1606Rollback.patch.txt](https://github.com/react-native-community/react-native-camera/files/2220689/PR1606Rollback.patch.txt)
I'll try a few more things too.

davidpricedev, 2018-07-23:
When testing on v1.1.3 and v1.1.4, I get a redbox error which also prevents the camera from working, but doesn't crash the app.
> Exception thrown while executing UI block: -[RNCamera updateFaceDetectionLandmarks:]: unrecognized selector sent to instance 0x129ffaeb0

davidpricedev, 2018-07-24:
The updateFaceDetectionLandmarks issue/error has something to do with the podfile structure. If I keep the 1.1.5 version of podspec/podfile then that error goes away. Patching just the ios folder back to 1.1.3 now works fine for me, and 1.1.4 as well.

davidpricedev, 2018-07-24:
It seems to be that [dea1dc8](https://github.com/react-native-community/react-native-camera/pull/1574) introduced the issue.

davidpricedev, 2018-07-24:
Further, it seems that adding this line to RNCamera.m is the source of the problem for me.
```
       self.session.sessionPreset = AVCaptureSessionPresetPhoto;
```

jgfidelis, 2018-07-24:
Nice work @davidpricedev 

@almost would you kindly give an insight on what could be done to fix this? Maybe change that line to another moment?

davidpricedev, 2018-07-24:
If I take the latest off of master right now and comment out that line, the app crash goes away for me.

KryDos, 2019-01-16:
by commenting out the line mentioned by @davidpricedev here - https://github.com/react-native-community/react-native-camera/issues/1613#issuecomment-407547670 I was able to get rid of that black screen issue. 

Not

PR #1618 (2018-06-08): fix(picture-size): create None default value
fix front camera problems reported on #1613

Closes #1613
closed 2019-06-08 by stale[bot]