react-native-camera/react-native-camera #1616: Cannot add output <AVCaptureMetadataOutput: 0x11ef05410> to capture session <AVCaptureSession: 0x122649600 [AVCaptureSessionPresetHigh]> because more than one output of the same type is unsupported

xwartz, 2018-06-07:
### Which implementation are you using

*RNCamera*

### Crash Stacktrace
[image]

### Environment
- **React Native version**:  0.51.0
- **React Native platform + platform version**: Android 5/6/7/8

### react-native-camera
**Version**: 1.1.1

rajivshah3, 2018-06-11:
+1 but slightly different stacktrace:

```
NSInvalidArgumentException *** -[AVCaptureSession addOutput:] Cannot add output <AVCaptureMetadataOutput: 0x1c4637940> to capture session <AVCaptureSession: 0x1c0201bb0 [AVCaptureSessionPresetHigh]> because more than one output of the same type is unsupported 
    Frameworks/CoreFoundation.framework/CoreFoundation ___exceptionPreprocess
    /usr/lib/libobjc.A.dylib _objc_exception_throw
    Frameworks/AVFoundation.framework/AVFoundation -[AVCaptureSession addOutput:]
    /Users/vagrant/git/src/mobile/node_modules/react-native-camera/ios/RN/RNCamera.m:669:13 -[RNCamera _setupOrDisableMetadataOutput]
    /Users/vagrant/git/src/mobile/node_modules/react-native-camera/ios/RN/RNCamera.m:659:5 -[RNCamera setupOrDisableBarcodeScanner]
    /Users/vagrant/git/src/mobile/node_modules/react-native-camera/ios/RN/RNCameraManager.m:182:5 -[RNCameraManager set_barCodeScannerEnabled:forView:withDefaultView:]
    /Users/vagrant/git/src/mobile/node_modules/react-native/React/Views/RCTComponentData.m:108:5 -[RCTComponentData callCustomSetter:onView:withProp:isShadowView:]
    /Users/vagrant/git/src/mobile/node_modules/react-native/React/Views/RCTComponentData.m:229:7 __49-[RCTComponentData createPropBlock:isShadowView:]_block_invoke_2
    /Users/vagrant/git/src/mobile/node_modules/react-native/React/Views/RCTComponentData.m:356:5 __37-[RCTComponentData setProps:forView:]_block_invoke
    Frameworks/CoreFoundation.framework/CoreFoundation -[__NSDictionaryM enumerateKeysAndObjectsWithOptions:usingBlock:]
    /Users/vagrant/git/src/mobile/node_modules/react-native/React/Views/RCTComponentData.m:355:3 -[RCTComponentData setProps:forView:]
    /Users/vagrant/git/src/mobile/node_modules/react-native/React/Modules/RCTUIManager.m:971:7 __50-[RCTUIManager createView:viewName:rootTag:props:]_block_invoke
    /usr/lib/system/libdispatch.dylib __dispatch_call_block_and_release
    /usr/lib/system/libdispatch.dylib __dispatch_client_callout
    /usr/lib/system/libdispatch.dylib __dispatch_main_queue_callback_4CF$VARIANT$mp
    Frameworks/CoreFoundation.framework/CoreFoundation ___CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__
    Frameworks/CoreFoundation.framework/CoreFoundation ___CFRunLoopRun
    Frameworks/CoreFoundation.framework/CoreFoundation _CFRunLoopRunSpecific
    PrivateFrameworks/GraphicsServices.framework/GraphicsServices _GSEventRunModal
    Frameworks/UIKit.framework/UIKit _UIApplicationMain
    appName/main.m:16:12 main
    /usr/lib/system/libdyld.dylib _start
```

`react-native-camera`: v1.1.4
`react-native`: v0.51.0
`react`: v16.0.0

reganperkins, 2018-06-15:
im getting this as well but can't yet tell what is causing it

rajivshah3, 2018-06-16:
>React Native platform + platform version: Android 5/6/7/8

@xwartz I think this is an iOS issue, `NSInvalidArgumentException` sounds like an iOS class. What iOS versions are you seeing it on?

xwartz, 2018-06-19:
@rajivshah3  Yes, it's iOS issue, it occur on 11.1.0 - 11.3.1

MannyFM, 2018-07-08:
+1.
```
"react": "16.3.1",
"react-native": "~0.55.2",
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera",
```
```
2018-07-08 21:29:32.096059+0600 ReactiveMuseum[249:12251] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureSession addOutput:] Cannot add output <AVCaptureMetadataOutput: 0x1c4429920> to capture session <AVCaptureSession: 0x1c020ad80 [AVCaptureSessionPresetPhoto]> because more than one output of the same type is unsupported'
*** First throw call stack:
(0x1861ff164 0x185448528 0x18bb6fcf0 0x1051619a0 0x105161718 0x10515ef28 0x105a912cc 0x105a9128c 0x105a9ff80 0x105a947ec 0x105aa0f6c 0x105aa8020 0x185e26f1c 0x185e26b6c)
libc++abi.dylib: terminating with uncaught exception of type NSException
2018-07-08 21:40:14.830417+0600 ReactiveMuseum[249:15129] XPC connection interrupted
```

svetoslav0625, 2018-07-30:
+1
I got this error on iOS
"react": "16.3.1",
"react-native": "0.55.4",
"react-native-camera": "^1.1.4",

SimonErm, 2018-08-01:
Can you test if this fix this? https://github.com/SimonErm/react-native-camera/tree/fix/multipleOutputOfSameType

eamonwhiter73, 2018-08-28:
@SimonErm I'm just curious...im newbish with github...how would i go about testing ur fork? just uninstall react-native-camera, and then install your version branch normally with `npm install`?

eamonwhiter73, 2018-08-29:
@SimonErm I'm just curious...im newbish with github...how would i go about testing this? just uninstall react-native-camera, and then install your fork?

eamonwhiter73, 2018-08-29:
@dmitryn hod does the reference issue relate?

SimonErm, 2018-08-29:
@eamonwhiter73 use "react-native-camera": "git+https://git@github.com/SimonErm/react-native-camera#multipleOutputOfSameType" instead of a version and run npm install

eamonwhiter73, 2018-08-29:
@SimonErm thanks ill try

eamonwhiter73, 2018-08-29:
@SimonErm I am getting `error: pathspec 'multipleOutputOfSameType' did not match any file(s) known to git.` when trying to run `npm install` and in your repo on git it says `Cannot retrieve the latest commit at this time.` Are there any commits I can retrieve?

SimonErm, 2018-08-29:
@eamonwhiter73 oh sorry, it has to be: `"react-native-camera": "git+https://git@github.com/SimonErm/react-native-camera#fix/multipleOutputOfSameType"`

yurnery, 2018-09-19:
@SimonErm hi, can you tell me which commit you fix this issue, I got no information from your commit list

theohdv, 2018-11-09:
@SimonErm could you explain your fix please ? I'll try it but i'd like to know what went wrong

AshokICreate, 2018-12-21:
```
iPhone 7 (D101AP)
Name | iOS
-- | --
Version | 11.4 (15F79)
```

```
NSInvalidArgumentException: Application threw exception NSInvalidArgumentException: *** -[AVCaptureSession addOutput:] Cannot add output <AVCaptureMetadataOutput: 0x1c042f880> to capture session <AVCaptureSession: 0x1c02078b0 [AVCaptureSessionPresetPhoto]> because more than one output of the same type is unsupported Originated at or in a subcall of GMSx_AbslInternalSpinLockWake
--
```

n1ru4l, 2019-01-08:
@AshokICreate Can you please provide a step by step guide to reproduce this issue with the example app in this repository?

AshokICreate, 2019-02-13:
```
- (void)_setupOrDisableMetadataOutput
{
    if ([self isReadingBarCodes] && (_metadataOutput == nil || ![self.session.outputs containsObject:_metadataOutput])) {
        AVCaptureMetadataOutput *metadataOutput = [[AVCaptureMetadataOutput alloc] init];
        if ([self.session canAddOutput:metadataOutput]) {
            NSLog(@"Stack trace : %@",[NSThread callStackSymbols]);
            [metadataOutput setMetadataObjectsDelegate:self queue:self.sessionQueue];
            @try {
                [self.session addOutput:metadataOutput];
            }
            @catch (NSException *exception) {
                NSLog(@"ashok the exception is %@", exception);
                NSLog( @"ashok Name: %@", exception.name);
                NSLog( @"ashok Reason: %@", exception.reason);
            }
            @finally {
                NSLog(@"Finally condition");
            }
            [self.session addOutput:metadataOutput];
            self.metadataOutput = metadataOutput;
        }
    } else if (_metadataOutput != nil && ![self isReadingBarCodes]) {
        [self.session removeOutput:_metadataOutput];
        _metadataOutput = nil;
    }
}
```

AshokICreate, 2019-02-13:
I changed to below code.

`- (void)_setupOrDisableMetadataOutput
{
    if ([self isReadingBarCodes] && (_metadataOutput == nil || ![self.session.outputs containsObject:_metadataOutput])) {
        AVCaptureMetadataOutput *metadataOutput = [[AVCaptureMetadataOutput alloc] init];
        if ([self.session canAddOutput:metadataOutput]) {
            @try {
                if(![self.session.outputs containsObject:metadataOutput]) {
                    if(_metadataOutput != nil && [self.session.outputs containsObject:_metadataOutput]) {
                        [self.session removeOutput:metadataOutput];
                    } else {
                        [metadataOutput setMetadataObjectsDelegate:self queue:self.sessionQueue];
                        [self.session addOutput:metadataOutput];
                        self.metadataOutput = metadataOutput;
                    }
                    
                }
            }
            @catch (NSException *exception) {
                NSLog(@"exception is %@", exception);
            }
        }
    } else if (_metadataOutput != nil && ![self isReadingBarCodes]) {
        [self.session removeOutput:_metadataOutput];
        _metadataOutput = nil;
    }
}`

but below code is getting executed twice by some thread
```
 [metadataOutput setMetadataObjectsDelegate:self queue:self.sessionQueue];
                        [self.session addOutput:metadataOutput];
                        self.metadataOutput = metadataOutput;
```

is this because of 
```
dispatch_async(self.sessionQueue, ^{
//some code
})
```

AshokICreate, 2019-02-14:
https://github.com/n1ru4l anything u can help?

AshokICreate, 2019-02-14:
> @AshokICreate Can you please provide a step by step guide to reproduce this issue with the example app in this repository?

@n1ru4l please find above comments

7772, 2019-04-30:
Is there any update here?

closed 2019-07-06 by stale[bot]