react-native-camera/react-native-camera #1624: Module ''react-native'' has no exported member 'ViewProperties'

pimuzzo, 2018-06-09:
Using react-native-camera > `1.0.2` I have this issue when launching yarn build on my project:

```
No valid rules have been specified
node_modules/react-native-camera/types/index.d.ts(13,10): error TS2305: Module ''react-native'' has no exported member 'ViewProperties'.
error Command failed with exit code 2.
```

jgfidelis, 2018-06-11:
please try newer versions

closed 2018-06-11 by sibelius

pimuzzo, 2018-06-13:
As previously written I tried the new versions, but the only one that works correctly is 1.0.1 (or less).
At the moment I'm trying to debug the problem, I'll update the ticket if I find out something.
If you have any advice I'll try it

pimuzzo, 2018-06-14:
Fixed, I write two lines in case it happens to someone else.
I'm writing an extension for reactxp and I noticed that it included a custom version of react-native types. removing it and installing the types from npm now everything works perfectly, sorry.