react-native-camera/react-native-camera #1626: Error: camera is not running

taomanwai, 2018-06-10:
Hi all,

I am using RNCamera takePictureAsync(), it works in iOS but failed in Android, any ideas?

### Steps to reproduce
1. Add RNCamera in page
2. Run const data = await this.camera.takePictureAsync({ quality: 0.5, base64: true });
3. "Error: camera is not running" exception occurred 

### Expected behaviour
Capture camera image as data variable

### Actual behaviour
"Error: camera is not running" exception occurred 

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: Android 7.1.0

### react-native-camera
**Version**: "master" (RNCamera is using)

taomanwai, 2018-06-15:
To n1ru4l, what more information u want?

massimopibiri, 2018-06-18:
The same problem here. The camera sometimes works when the application is rebuilt. But he stops working suddenly for no reason, even when nothing has changed in the code.

massimopibiri, 2018-06-18:
Found the solution :
I read in stackoverflow that the problem comes from the router (I use react-native-router-flux). I don't know if this is true or not. I didn't understand it. In my case, everything started to work as soon as I removed the following code from inside the RNCamera:

```
          {({ camera, status }) => {
            if (status !== 'READY') return <PendingView />;
            return (
              <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
                <TouchableOpacity onPress={() => this.takePicture(camera)} style={styles.capture}>
                  <Text style={{ fontSize: 14 }}> SNAP </Text>
                </TouchableOpacity>
              </View>
            );
          }}
```
And I simply put the button without verifying if the camera is 'ready':

```
              <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
                <TouchableOpacity onPress={() => this.takePicture(camera)} style={styles.capture}>
                  <Text style={{ fontSize: 14 }}> SNAP </Text>
                </TouchableOpacity>
              </View>
```
and it works perfectly.

massimopibiri, 2018-06-18:
Sorry... without the "camera" inside this.takePicture()

geisterfurz007, 2018-07-02:
Appears my issue is a duplicate of this https://github.com/react-native-community/react-native-camera/issues/1426#issuecomment-381566152

telmen, 2018-07-04:
This is happening on Android 4.4.2

shaksi, 2018-09-02:
had similar issue where the camera is loading on iOS and not on android. 
following @massimopibiri advice adapted the code as follows

`{({ camera, status }) => {
//check if camera is ready and platform is not android
//if android assume its ready
            if (status !== 'READY' && Platform.OS === "android") return <PendingView />;
            return (
              <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
                <TouchableOpacity onPress={() => this.takePicture(camera)} style={styles2.capture}>
                  <Text style={{ fontSize: 14 }}> SNAP </Text>
                </TouchableOpacity>
              </View>
            );
          }}`

zvs001, 2018-10-09:
I had this issue with `react-navigation` router.
Camera stops working after I changing page
This code solved it for me:
``` 
import { withNavigationFocus } from 'react-navigation' 
```

```
render() {
    const { isFocused } = this.props
    return (
       <View>
         { isFocused && <RNCamera  ... />  }
       </View
     )
}
```
``` 
export default withNavigationFocus(Camera) 
```

closed 2019-06-08 by stale[bot]

brsaylor2, 2019-11-08:
I'm using `react-native-navigation` and was seeing this bug after displaying a modal over the camera. I solved it by passing the [`modalPresentationStyle: 'overCurrentContext'`](https://wix.github.io/react-native-navigation/#/docs/styling?id=common-options) option to `showModal`.

blackstalk, 2022-09-15:
I was getting the "Camera not running" error when i changed screens. So for functional components instead of `withNavigationFocus(Camera)` use this method:
```

import { useIsFocused } from '@react-navigation/native';

function MyComponent() {
  const isFocused = useIsFocused()
  
  return (
    <View>
      { isFocused && <RNCamera />  }
    </View>
  )
}
```