react-native-camera/react-native-camera #1631: undefined is not an object (evaluating 'CameraManager.aspect') during running app

sarikabhimre18, 2018-06-12:


sibelius, 2018-06-12:
your linking is wrong

check docs, link right and rebuild

closed 2018-06-12 by sibelius

connercms, 2018-06-18:
+1.  I think most of us are following the docs exactly and experiencing this issue, so it would be nice to get a better explanation than "your link is wrong" that is posted in every thread involving this issue. What do you mean "your linking is wrong"?

ShahoG, 2018-06-26:
Have the exact same issue,  installed and linked according to docs but get this instead:

> rnpm-install ERR! Something went wrong while linking. Error: Expected "/* Begin ", "/* End ", "\"", or [A-Za-z0-9_.] but "<" found.
Please file an issue here: https://github.com/facebook/react-native/issues

imspikey, 2018-07-05:
I solved this issue buy cleaning and rebuilding with android studio 👍

anniewey, 2018-09-24:
Run `react-native-link` didn't link the library completely.
Manually add `libRNCamera.a` to **Linked Frameworks and Libraries** solve the problem for me

n1ru4l, 2019-02-01:
@riyadshauk You can either do that or correctly link the library 😇