react-native-camera/react-native-camera #1634: flashMode.torch not turning on light RNCamera

mcoder14, 2018-06-13:
Using *RNCamera* 

### Steps to reproduce
<RNCamera
					ref={cam => {
						this.camera = cam;
					}}
					style={this.getStyle().preview}
					type={RNCamera.Constants.Type.back}
					flashMode={RNCamera.Constants.FlashMode.torch}
					onFocusChanged={() => { }}
					onZoomChanged={() => { }}
					defaultTouchToFocus
					mirrorImage={false}
					barCodeTypes={[
						RNCamera.Constants.BarCodeType.aztec,
						RNCamera.Constants.BarCodeType.code39,
						RNCamera.Constants.BarCodeType.code39mod43,
						RNCamera.Constants.BarCodeType.code93,
						RNCamera.Constants.BarCodeType.code128,
						RNCamera.Constants.BarCodeType.datamatrix,
						RNCamera.Constants.BarCodeType.ean13,
						RNCamera.Constants.BarCodeType.ean8,
						RNCamera.Constants.BarCodeType.interleaved2of5,
						RNCamera.Constants.BarCodeType.itf14,
						RNCamera.Constants.BarCodeType.pdf417,
						RNCamera.Constants.BarCodeType.qr,
						RNCamera.Constants.BarCodeType.upce,
					]}
					onBarCodeRead={barcodeReadFunction}
				>

### Expected behaviour
The torch mode should turn on the flashlight and it should stay on

### Actual behaviour
The flash flickers twice and then does not turn on

### Environment
-  "react-native": "0.55.3",
-  iOS 11.3

### react-native-camera
      "react-native-camera": "^1.1.2",

jlemm45, 2018-06-13:
Torch mode needs to be set after the camera has already loaded so essentially try using a state variable tied to a button toggle of some sort to control which flashMode is on. 

Example:
`flashMode={torchOn ? Camera.Constants.FlashMode.torch : Camera.Constants.FlashMode.off}`

n1ru4l, 2018-06-13:
We should improve this!

zhangxinagjunHJH, 2018-07-07:
Have you solved the problem already？

zhangxinagjunHJH, 2018-07-09:
I rebuild project from xcode, it worked!

joaaomanooel, 2019-03-14:
I also have this problem on Android and tried these solutions, but it did not work

GotiMehta, 2019-03-26:
@jordanmkoncz  I tried your solution however it is only working in Android and not in iOS. Bit strange but could not successful. Torch just blinks once and goes off.

I followed below reference for it:
[https://github.com/react-native-community/react-native-camera/issues/1626#issuecomment-402609694](url)

Does any body have idea what causes the issue?

smeevil, 2019-05-10:
I have the exact same thing as @GotiMehta on iOS.

GotiMehta, 2019-05-16:
@smeevil 

Use [react-native-torch](https://www.npmjs.com/package/react-native-torch). It works in both platforms.

closed 2019-07-22 by stale[bot]