react-native-camera/react-native-camera #1639: com.google.cameraview: should be a subtree

eqyiel, 2018-06-18:
It's not easy to tell what has changed between the upstream [cameraview](https://github.com/google/cameraview) and the one in this repo since it was added [all in one shot with modifications](https://github.com/react-native-community/react-native-camera/commit/d11ed31917c26df151b4fb46ab166d2921a9ac99#diff-3a45c66f80f8ef3e15099f554b5dbdb8).

If it's necessary to make local modifications, I think it should be subtree-d into this repo instead so that

- it's possible to tell what changed between the local one and upstream without cloning both repos and using `diff -r`
- it's possible to pull changes (bug fixes, improvements) from upstream

sibelius, 2018-06-18:
Cameraview has been stopped for a long time

eqyiel, 2018-06-18:
@sibelius I don't understand, the last commit was about a year ago? https://github.com/google/cameraview/commit/3eaeac09a7f084463a94850c763d21fdb4c18fce

sibelius, 2018-06-18:
It has not been maintained anymore

eqyiel, 2018-06-18:
@sibelius I see you have some experience with that: https://github.com/google/cameraview/pull/218

And that this exists: https://github.com/react-native-community/cameraview

Does anyone know the reason for not continuing to use that fork?

alexshikov, 2018-06-26:
Hey guys, I would like to adjust **cameraview** library as well. Seems there that's not so easy to find a fresh (and probably supported) repo.

@sibelius is https://github.com/react-native-community/cameraview something that's planned to be supported and I should fork and (and contribute PR's if any)?
Or that repo won't be supported also, and I should clone current repo and use android camera source code from it?

sibelius, 2018-06-26:
I think we can improve cameraview on https://github.com/react-native-community/cameraview

then bring this changes here and test using RNCamera

does this workflow works for you?

alexshikov, 2018-06-26:
Sure! That sounds like the best flow for me.

alexshikov, 2019-01-31:
@ethaizone just keep in mind, the original CameraView from Google repo has a few bugs and it's not supported anymore by Google (or anybody).

https://github.com/react-native-community/cameraview has a few fixes comparing to original repo.

johcam12, 2019-03-09:
What about this one?
https://github.com/natario1/CameraView

wenq1, 2019-03-09:
Maybe this one is more recent?
https://github.com/natario1/CameraView

sibelius, 2019-03-09:
we going to accept a pull request that transform the current camera view code inside this repo to a subtree inside this repo

closed 2019-06-07 by stale[bot]