react-native-camera/react-native-camera #1642: GPS Coordinates in exif

Camochase, 2018-06-18:
I am using RNCamera and I cannot seem to find a way to include GPS coordinates in the exif data and there doesn't seem to be any documentation on this either. If somebody could provide an example or let me know if this is possible that would be great.

hampelm, 2018-07-31:
We used to be able to use `metadata.location`: https://github.com/react-native-community/react-native-camera/blob/master/docs/RCTCamera.md#captureoptions-promise 

But that seems to no longer be supported. Not seeing promising results when searching the RNCamera source

albullington, 2018-10-18:
I'm also looking for a way to save location data while taking photos with react-native-camera. Did anyone find a fix for this?

manel27, 2019-01-08:
Someone? Help.

closed 2019-06-24 by stale[bot]

cristianoccazinsp, 2019-11-10:
This PR https://github.com/react-native-community/react-native-camera/pull/2573 adds code so that EXIF is preserved on iOS. However, neither Android nor iOS automatically set the GPS tags to the captured media. Looks like this needs to be manually added afterwards, or on the same capture event.

I've been thinking about being able to pass additional EXIF data to the takePictureAsync call (Android already attempts to implement this when not using skipProcessing (https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java#L137) but it is not documented anywhere.

Now that the above PR allows for metadata/exif to be written to the final image, we could implement the same for both iOS and Android. Sadly, this would mean that people trying to add GPS tags to their pictures would first need to get the GPS values from the JS side, learn the EXIF tag name, and then pass it to the camera call.