react-native-camera/react-native-camera #1646: Android crash onBarCodeReadEvent

reganperkins, 2018-06-18:
RNCamera

### Steps to reproduce
1. have version 1.1.4 react-native-camera installed. one one device (Android 6.0.1 Nexus 5) the error only occurs when in production mode (code is minified) on the Android device 8.1.0 it occurs all the time.
2. launch react-native-camera and include the following

```
this.barCodeTypes = Platform.select({
	ios: ['org.iso.PDF417'],
	android: ['PDF_417']
});

<RNCamera
	ref={(cam) => { this.camera = cam; }}
	onBarCodeRead: this._onBarCodeRead,
	barCodeTypes: this.barCodeTypes,
/>
```
3. scan pdf417 barcode (ie a drivers license)

### Expected behaviour
App should execute onBarCodeRead and not crash the app.

### Actual behaviour
onBarCodeRead gets triggered with the correct data from the pdf417 barcode but crashes with `java.lang.NullPointerException: Attempt to invoke virtual method 'float com.google.zxing.ResultPoint.getX()' on a null object reference` once you restart the app you can see that the data had been sent to the server correctly from the onBarCodeRead method (ie onBarCodeRead logic was completed).

### Environment
- **React Native version**:  0.55.4
- **React Native platform + platform version**: Android 6.0.1 and Android 8.1.0

### react-native-camera
1.1.4 and 1.1.5

*** ERROR Message ***
```
 E/AndroidRuntime: FATAL EXCEPTION: mqt_js
    Process: **.***.*****, PID: 21239
    java.lang.NullPointerException: Attempt to invoke virtual method 'float com.google.zxing.ResultPoint.getX()' on a null object reference
        at org.reactnative.camera.events.BarCodeReadEvent.serializeEventData(BarCodeReadEvent.java:71)
        at org.reactnative.camera.events.BarCodeReadEvent.dispatch(BarCodeReadEvent.java:58)
        at com.facebook.react.uimanager.events.EventDispatcher$DispatchEventsRunnable.run(EventDispatcher.java:351)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
        at android.os.Looper.loop(Looper.java:164)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
        at java.lang.Thread.run(Thread.java:764)
06-18 11:40:18.804 916-20760/? W/ActivityManager:   Force finishing activity 
```

haggholm, 2018-06-18:
The same problem occurs in the same app (I am on the same team as @reganperkins) with `react-native-camera` 1.1.5, which has a different version of `zxing` (3.2.1→3.3.0).

haggholm, 2018-06-18:
The same issue occurs whether Proguard is enabled or not.

smerat, 2018-06-18:
any help with this would be appreciated.

hardcodet, 2018-06-19:
+1
Seeing this in DEV occasionally. There's no pattern as far as I can tell - sometimes it works, sometimes it doesn't.

SimonErm, 2018-06-28:
Can you try https://github.com/SimonErm/react-native-camera/tree/fix/NullPointerExceptionOnBarcodeRead ? If it works i'll do a PR. I don't know why the resultPoints contain a null point, but this should prevent the crashes.

closed 2018-06-29 by SimonErm