react-native-camera/react-native-camera #1648: Argument list too long: recursive header expansion failed

JindiHu, 2018-06-19:
React native camera 1.1.4
i followed the instruction and manually link for ios for no face detection setup. 
but when i build the project, i got this error 

Argument list too long: recursive header expansion failed

can someone share the solution?

VicFrolov, 2018-06-19:
I opened a ticket about this a few months ago, and it was closed stating it was fixed, but this was not fixed for me.

You need to go to Xcode -> Libraries -> RNCamera.xcodeproj -> Framework Search Paths

this works for me: `$(SRCROOT)/../../ios` and I need to change it from `$(SRCROOT)/../../../ios`

vgeorge, 2018-06-20:
@JindiHu Did you look into issue #1407? I managed to install without face detection by installing via CocoaPods, as explained [here](https://github.com/react-native-community/react-native-camera/issues/1407#issuecomment-397683592).

sibelius, 2019-03-18:
check this https://github.com/react-native-community/react-native-camera/issues/1407

closed 2019-03-18 by sibelius