react-native-camera/react-native-camera #1652: when orientation is locked programmatically, recorded video is sometimes rotated

manask88, 2018-06-20:
### Which implementation are you using

RNCamera

### Steps to reproduce
1.  Lock the screen orientation to portrait mode programatically
2. Mount the camera component
3. start recording

### Expected behaviour
If the video has been recorded with the physical orientation of the phone in landscape or portrait mode, those videos should have the correct orientation when they are played.

### Actual behaviour
- When the orientation is locked to portrait programmatically and the phone is placed in landscape physical orientation, the recorded video comes rotated.
- When the orientation is locked to landscape programmatically and the phone is placed in portrait physical orientation, the recorded video comes rotated.

### Environment
- **React Native version**:  0.55.4
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master": master

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

manask88, 2018-06-25:
@sibelius , please let me know your thoughts about this so I can start working on a potential solution

n1ru4l, 2018-12-12:
Possible related to #1725

closed 2019-06-08 by stale[bot]