react-native-camera/react-native-camera #1655: [Android] Could not find method implementation() for arguments [com.google.zxing:core:3.2.1]

longpq18, 2018-06-21:
Could not find method implementation() for arguments [com.google.zxing:core:3.2.0] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

My react-native: 
react-native-cli: 2.0.1
react-native: 0.53.3

sibelius, 2018-06-21:
can u post the full stacktrace?

longpq18, 2018-06-21:
myproject/node_modules/react-native-camera/android/build.gradle' line: 56

* What went wrong:
A problem occurred evaluating project ':react-native-camera'.
> Could not find method implementation() for arguments [com.google.zxing:core:3.2.1] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

punksta, 2018-06-24:
upgrade your gradle and android-gradle-plugin to resolve issue: https://stackoverflow.com/a/46619296/2795206

closed 2018-06-24 by punksta