react-native-camera/react-native-camera #1656: Out of Memory Bitmap Pool error

sharmad1205, 2018-06-21:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

I am using RNCamera 1.1.4   
and got this error

Fatal Exception: java.lang.OutOfMemoryError
Failed to allocate a 51121164 byte allocation with 15851520 free bytes and 41MB until OOM
dalvik.system.VMRuntime.newNonMovableArray (VMRuntime.java)
android.graphics.Bitmap.createBitmap (Bitmap.java:915)
com.facebook.imagepipeline.memory.BitmapPool.alloc (BitmapPool.java:55)
com.facebook.imagepipeline.memory.BitmapPool.alloc (BitmapPool.java:30)
com.facebook.imagepipeline.memory.BasePool.get (BasePool.java:260)

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?
Check usage with Expo - https://github.com/react-native-community/react-native-camera/blob/master/docs/Expo_Usage.md
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sharmad1205, 2018-06-21:
Got two more crashes as below
1. 
Caused by java.lang.OutOfMemoryError
Failed to allocate a 1050476 byte allocation with 348752 free bytes and 340KB until OOM
java.lang.StringFactory.newStringFromChars (StringFactory.java)
android.util.Base64.encodeToString (Base64.java:456)
org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground (ResolveTakenPictureAsyncTask.java:130)

2. 
Caused by java.lang.OutOfMemoryError
Failed to allocate a 51121164 byte allocation with 16776312 free bytes and 46MB until OOM
dalvik.system.VMRuntime.newNonMovableArray (VMRuntime.java)
android.graphics.BitmapFactory.decodeByteArray (BitmapFactory.java:636)
org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground (ResolveTakenPictureAsyncTask.java:83)

SimonErm, 2018-09-26:
Are you useing mirrorImage or fixOrientation?

n1ru4l, 2019-01-22:
Duplicate of #1914

closed 2019-01-22 by n1ru4l