react-native-camera/react-native-camera #1658: onTextRecognized method Example Please

devbasil, 2018-06-21:
### Which implementation are you using
RNCamera master branch

Kindly Provide an example I cant get it to work with the code below

**My Code**
<RNCamera
              ref={ref => {
                this.camera = ref;
              }}
              style={styles.preview}
              type={RNCamera.Constants.Type.back}
              autoFocus={RNCamera.Constants.AutoFocus.on}
              permissionDialogTitle={'Permission to use camera'}
              permissionDialogMessage={'We need your permission to use your camera to Scan'}
              onTextRecognized={this.props.getCameraIsActive === true ? this.onTextRecognized : null}
              >
              <TextFinder width={220} height={220} borderColor="#FFFFFF" borderWidth={2} />
          </RNCamera>

**My Method** 
onTextRecognized = (e) => { 
...................................
this.props.saveCameraIsActive(false);
Vibration.vibrate(500)
.................................................
.............................................
}

### Expected behaviour
Should return actual text blocks 

### Actual behaviour
I get empty textBlocks array 

### Environment
- **React Native version **: 0.55.4
- **React Native platform + platform version**: Android 7.0, etc

### react-native-camera
**Version**:  "master branch"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

closed 2018-06-22 by devbasil

sibelius, 2018-06-22:
could u add an example do /docs folder?

dileepbolisetti, 2018-11-11:
yeah add an example please.