react-native-camera/react-native-camera #1662: [iOS] when orientation is locked using the lock orientation menu option, recorded video has the wrong orientation

manask88, 2018-06-25:
### Which implementation are you using

RNCamera

### iOS

### Steps to reproduce
1.  Lock the screen orientation to portrait mode using the iOS lock orientation option
2. Mount the camera component
3. start recording

### Expected behaviour
If the video has been recorded with the physical orientation of the phone in landscape or portrait mode, those videos should have the correct orientation when they are played.

### Actual behaviour
- When the orientation is locked using the iOS menu and the phone is in portrait phisycal orientation, 
- When the orientation is locked using the iOS menu and the phone is in landscape physical orientation, 

### Environment
- **React Native version**:  0.55.4
- **React Native platform + platform version**:  iOS 11

### react-native-camera
**Version**: npm version or "master": master

In order to make this work some changes in the native implementation need to be done. In https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L451 . [UIDevice currentDevice] orientation] is used. This method to get the device orientation has some limitations since it doesn't get the real device orientation when the device is locked. That orientation is okay to use for the video preview, but in order to set the right orientation to the video output, a different method needs to be used.

A suggested approach is to use CoreMotion. This is approach is suggested in:
https://stackoverflow.com/questions/39051033/how-to-detect-orientation-if-screen-is-locked-in-portrait-mode
http://www.prettyitgirl.com/2015/11/detect-device-rotation-even-when-rotation-lock-is-on.html. 
and was also used in the old react-native-camera implementation
https://github.com/react-native-community/react-native-camera/blob/master/ios/RCT/RCTSensorOrientationChecker.m

I'd love to send a PR with this solution. However, I want to understand if you guys would be willing to accept my changes.

manask88, 2018-06-25:
@sibelius , please let me know your thoughts about this so I can start working on a potential solution

pribeh, 2018-07-22:
We're experiencing this issue as well with certain users of iOS 11.2.6 and iPhone 8. Looking into your suggested solutions

jgfidelis, 2018-07-23:
@manask88 sorry for the late response, if you would send the pr and test it, of course we would accept it.