react-native-camera/react-native-camera #1663: permissionDialogTitle and permissionDialogMessage ignored

zedtux, 2018-06-26:
### Which implementation are you using

RNCamera

### Steps to reproduce
1. Build an app based on [the documentation Usage](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#usage)
2. Run the app on an Android (7.1.2) phone (Xiaomi Redmi)

### Does it work with Expo Camera?

I'm not using Expo.

### Expected behaviour

When running the app the first time, it should ask the permission to access the camera using what has been defined in the `permissionDialogTitle` and `permissionDialogMessage`.

### Actual behaviour

It show a default message asking if the user allow the app name to take picture and record videos.

### Environment

- **React Native version**: 0.55.4
- **React Native platform + platform version**: Android 7.1.2

### react-native-camera

**Version**: 1.1.4

jgfidelis, 2018-06-27:
Please provide relevant code snippet, like your `<RNCamera>`

zedtux, 2018-06-28:
Okay .. closed like that ... 🤦‍♂️. What a useless behaviour @sibelius 👍