react-native-camera/react-native-camera #1665: I get Cannot initialize a variable of type 'CFDictionaryRef' when building RNCamera in Xcode

ingviso, 2018-06-28:
### Which implementation are you using

RNCamera

### Steps to reproduce
1. yarn add react-native-camera
2. build app wich import RNCamera

### Expected behaviour
I should be able to build the app with RNCamera imported inside Xcode using iPhone 6s with iOS 11,4 and run the app.

### Actual behaviour
When building RNCamera in Xcode an error occurs inside:
Project > Libraries > RNCamera.xcodeproj > RN > RNImageUtils.m > +updatePhotoMetadata:withAdditionalData:inResponse

On line 98 this error comes up
Cannot initialize a variable of type 'CFDictionaryRef' (aka 'const __CFDictionary *') with an rvalue of type 'CFTypeRef _Nullable' (aka 'const void *')

Code in line 98: 
`CFDictionaryRef exifAttachments = CMGetAttachment(imageSampleBuffer, kCGImagePropertyExifDictionary, NULL);`

### Environment
Platform:  iOs 11,4
"react-native": "^0.55.2",
"react-native-camera": "^1.1.4"

jgfidelis, 2018-06-29:
what xcode version are you using?

uriklar, 2019-05-28:
Hi @ingviso i'm getting this too. Any chance you remember the solution?

closed 2019-08-03 by stale[bot]