react-native-camera/react-native-camera #1669: Camera Not Appearing

SeamusY, 2018-06-29:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?
RNCamera
### Steps to reproduce
After following the documentation and the Upgrade Gradle File and doing all the react-native links.

`<View style={styles.container}>
                <RNCamera
                    ref={ref => { this.camera = ref; }}
                    style={styles.preview}
                    type={RNCamera.Constants.Type.back}
                    flashMode={RNCamera.Constants.FlashMode.on}
                    permissionDialogTitle={'Permission to use camera'}
                    permissionDialogMessage={'We need your permission to use your camera phone'}
                    barCodeTypes={[RNCamera.Constants.barCodeTypes.pdf417]}
                    onBarCodeRead={this.BarCodeRead.bind(this)}
                />
</View>`
### Does it work with Expo Camera?
Not sure how to use EXPO camera

### Expected behaviour
Camera Appear on the Screen

### Actual behaviour
Blank White Screen

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

SimonErm, 2018-06-29:
What styles do you use? Did you add permissions to Manifest?

SeamusY, 2018-06-29:
here are the following styles:
`import {
  StyleSheet,
  Dimensions
} from 'react-native';
export default styles = StyleSheet.create({
  container: {
    flex: 1,
    justifyContent: 'center',
    alignItems: 'center',
    backgroundColor: '#F5FCFF',
  },
  preview: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
    backgroundColor: 'transparent',
    height: Dimensions.get('window').width,
    width: Dimensions.get('window').width,
  },
  withLimit: {
    width: '75%'
  },
  rectangleContainer: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
    backgroundColor: 'transparent',
  },
  rectangle: {
    height: 250,
    width: 250,
    borderWidth: 2,
    borderColor: '#00FF00',
    backgroundColor: 'transparent',
  },
});`
Yes, all manifest has been added.. the gradle has been added so that every runs accordingly.

SimonErm, 2018-06-30:
I couln't run your Code because of `barCodeTypes={[RNCamera.Constants.barCodeTypes.pdf417]}` try `barCodeTypes={[RNCamera.Constants.BarCodeType.PDF417]}` instead. But i got a red screen and after the change it worked like expected.

SeamusY, 2018-07-05:
In the `build.gradle` in apps:
```
android {
    compileSdkVersion 27
    buildToolsVersion '27.0.3'

    defaultConfig {
        applicationId "com.login"
        minSdkVersion 16
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
    }

dependencies {
    implementation (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        exclude group: "com.android.support"
    }
    implementation ('com.google.android.gms:play-services-vision:12.0.1') {
        force = true
    }
    implementation "com.android.support:exifinterface:27.+"
    implementation "com.android.support:support-annotations:+"
    implementation "com.android.support:support-v4:+"
    implementation fileTree(dir: "libs", include: ["*.jar"])
    implementation ("com.android.support:appcompat-v7:27.1.0"){
        force = true
    }
    implementation "com.facebook.react:react-native:+"  // From node_modules
}```

SeamusY, 2018-07-05:
Both emulation and physical show the same issue.

SeamusY, 2018-07-05:
After playing around with the gandle, it worked finally.

closed 2018-07-05 by SeamusY

swaroopa94, 2018-08-25:
@SeamusY Hi I am also facing same issue can you tell me how to solve this
`dependencies {
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        exclude group: "com.android.support"
        
    }
    compile project(':react-native-permissions')
    compile project(':react-native-google-analytics-bridge')
    compile project(':react-native-vector-icons')
    compile project(':react-native-video')
    compile project(':react-native-push-notification')
    compile project(':react-native-orientation')
    compile project(':react-native-image-resizer')
    compile project(':react-native-android-permissions')
    compile project(':react-native-webrtc')
    compile(project(':react-native-maps')){
      exclude group: 'com.google.android.gms', module: 'play-services-base'
      exclude group: 'com.google.android.gms', module: 'play-services-maps'
    }

    compile project(':react-native-device-info')
    compile fileTree(dir: "libs", include: ["*.jar"])
  
    compile 'com.android.support:appcompat-v7:26.+'
    compile "com.facebook.react:react-native:+"  // From node_modules
    compile project(':react-native-image-picker')
    compile 'com.google.android.gms:play-services-base:+'
    compile 'com.google.android.gms:play-services-maps:+'
    compile fileTree(dir: "node_modules/react-native-device-info/android/libs", include: ["*.jar"])

}`

swaroopa94, 2018-08-27:
@SeamusY YEs I checked and other thing I want to say is

the onCameraReady function is returning undefined when using like this:

`<RNCamera
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.on}
          permissionDialogTitle={'Permission to use camera'}
          onCameraReady={(res)=>alert(res)}
          onMountError={(err)=>alert("err"+err)}
          permissionDialogMessage={'We need your permission to use your camera phone'}
        >
          
        </RNCamera>`

 Can you tell me why it is returning undefined?

SpQuyt, 2018-09-08:
Can you show me your build.gradle in project, in android folder;
And the gradle-wrapper.properties please?