react-native-camera/react-native-camera #1671: Camera shows BLACK SCREEN

Albert0405, 2018-06-30:
### Problem
Im using react-native-camera for scanning QR Codes and react-navigation for TabNavigation. I wanted to put the Camera in one of the Tabs i created in react-navigation unfortunately when i try to navigate on other tabs and then go back to camera tab, the camera became black

1. Camera is working fine
2. I press on Home Tab which contains nothing but text "Home"
3. I press back to QrScanner Tab to see if the camera still working but unfortunately it became black and cannot scan QR Code anymore

[image]

### Which implementation are you using
QR Code Scanner

### Does it work with Expo Camera?
Yes, Check it here and run in your device
https://snack.expo.io/rkPEdeSzX

### Expected behaviour
I just want the camera to be still working after i press on the Tabs on my TabNavigator

SimonErm, 2018-06-30:
Take a look at #1051 and #999. If the answers don't work for you, you can reopen the issue.

closed 2018-06-30 by SimonErm

Albert0405, 2018-07-01:
I fixed using react navigation isFocused and withNavigationFocus

Albert0405, 2018-07-09:
By the way im having a performance issue when using withNavigationFocus. Was there any other way to fix this?

dileepbolisetti, 2018-08-28:
@Albert0405 pls provide sample code i am unable to achieve it thanks in advance

Albert0405, 2018-08-29:
@dileepbolisetti  Here

```
import { withNavigationFocus } from "react-navigation";
import QRCodeScanner from "react-native-qrcode-scanner";
```

```
class QrCodeCamera extends Component {

      renderCamera() {
         const isFocused = this.props.navigation.isFocused();
         
         if (!isFocused) {
             return null;
         } else if (isFocused) {
             return (
                <QRCodeScanner />
             )
         }

      render() {
         return (
           <View style={{ flex: 1 }}>
              {this.renderCamera()}
           </View>
     }
}

export default withNavigationFocus(QrCodeCamera);
```

raphaelhovsepyan, 2018-09-06:
Don't Wrap `RNCamera` with other components, it should be on the top

munonn430, 2018-12-10:
can i have the custom marker source code?

Hirurgo, 2019-01-01:
> Don't Wrap `RNCamera` with other components, it should be on the top

Где ваши доказательства. RNCamera wrapped in View in official doc.

https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md

n1ru4l, 2019-01-08:
@raphaelho wrote
> Don't Wrap RNCamera with other components, it should be on the top

That is not true, you can wrap `RNCamera` into any component. (as @Hirurgo said).