react-native-camera/react-native-camera #1674: 'App has stopped' when I install the RNCamera component

Gmtas, 2018-07-02:
### ./build.gradle
``` 
buildscript {
    repositories {
        jcenter()
        google()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.0.1'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        google()
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }

        maven {
            url 'https://maven.google.com'
        }

        maven {
            url "https://jitpack.io"
        }

    }
}
```
 
### ./app/build.gradle

```
dependencies { 
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        exclude group: "com.android.support"
    }
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
}

```

### ~/MailApplication.java
```
...
import org.reactnative.camera.RNCameraPackage;
...
@Override
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(),
            new RNCameraPackage()
      );
    }
```

### ./gradle/wrapper/gradle-wrapper.properties
```
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists
distributionUrl=https\://services.gradle.org/distributions/gradle-4.1-all.zip
```

### ./settings.gradle

```
rootProject.name = 'CameraTest'

include ':react-native-camera'
project(':react-native-camera').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-camera/android')

include ':app'
```

### ./App.js

```
import React, { Component } from 'react';
import { StyleSheet, Text, TouchableOpacity, View } from 'react-native';
import { RNCamera } from 'react-native-camera';

export default class App  extends Component {
  render() {
    return (
      <View style={styles.container}>
        <RNCamera
            ref={ ref => {
              this.camera = ref;
            }}
            style = {{ flex:1 }}
            type={RNCamera.Constants.Type.back} 
        /> 
      </View>
    );
  }
```

### App Has Stopped
I have installed these modules and corrected many of the errors I have made. Eventually I tried to run the application but this time the program is shutting down while I switch to the screen where I installed the camera component. I do not understand where I'm doing wrong. can someone help me?

 
[image]

sibelius, 2018-07-02:
what is the android crash stacktrace?

Gmtas, 2018-07-03:
how can ı see the android crash stacktrace?

jgcmarins, 2018-07-03:
You can run device connected to your computer, with `adb logcat` command line and catch all logs.
Check this: https://developer.android.com/studio/command-line/logcat

Gmtas, 2018-07-03:
Thank you I got it...

the error is as follows

[image]

"java.lang.NoSuchMethodError: android.support.v4.view.ViewCompat.getDisplay"

The setting I made in practice is as above. Where could I have done wrong. In addition, do you think these settings are the best settings for the RNCamera component?

Gmtas, 2018-07-04:
no one has any idea?

Gmtas, 2018-07-04:
I solved my problem with [react-native-camera-kit](https://github.com/wix/react-native-camera-kit). it works great. I suggest it.

closed 2019-06-08 by stale[bot]