react-native-camera/react-native-camera #1677: Issues on QRcode read and Performance issues

SeamusY, 2018-07-05:
### Which implementation are you using
RNCamera
### Steps to reproduce
1. Create a stack using react-navigator, and on one page load the camera in full screen.
2. Start trying to Scan QR codes or move around (It should lag a lot)

### Does it work with Expo Camera?
No, because i am physically loading the camera on to the screen
### Expected behaviour
The camera shouldn't lag as much + being able to read the QRcode in about a 1-3 seconds

### Actual behaviour
The phone when moving lags, and when i get to the QRcode it has auto focus trying to figure out the data from the QR code. When rotating there is a chance of the QR code being read which then sends me to another react-navigation stack page. The chances so far of my scanning any QR code is about 1/10 chance of reading. oddly when holding horizontally i cannot read the QR code, however the performance is much better than holding it vertically.

### Environment
- **React Native version**: 0.56 (Before i was also using 0.55
- **React Native platform + platform version**:  Android 5.0.1

### react-native-camera
**Version**: Master

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

Sangrene, 2018-07-09:
Any fix ? Or is downgrading to a previous version working ?

SeamusY, 2018-07-10:
@Sangrene unfortunately, not with the current build (Master)... However, i did find something else interesting the "react-native-google-scanner" that was based off of "react-native-camera" performs slightly better. Although based of the three i used "react-native-barcode-scanner-google" performed the best with smooth and decent accuracy.

SeamusY, 2018-07-10:
I am not sure, but i believe something on the java side needs to be done, since performance is based upon the "native" element (Still playing around with it in my spare time).

Sangrene, 2018-07-10:
@SeamusY I also tried **react-native-qrcode-scanner** and yes, better performance overall but still not enough comparing to native QRCode apps... 
I gave a try to **react-native-barcode-scanner-google** but unfortunately ran into this issue : https://github.com/ekreutz/react-native-barcode-scanner-google/issues/29

Sadly I will surely need to debug this camera problem in java as well...

Udbhav12, 2018-10-17:
@SeamusY - I was facing the exact same problem as you while scanning QR codes/Barcodes, especially in Samsung Phones. I have included 'com.journeyapps:zxing-android-embedded:3.0.3@aar' in my build.gradle and developed a native module for scanning QR codes in android

closed 2019-06-08 by stale[bot]