react-native-camera/react-native-camera #1685: The necessity for gradle manual changes

taboca, 2018-07-08:
When attempting to build this project, right out of the box or following what seemed to be simple instructions, I have stumbled in the problem described at issue [1530](https://github.com/react-native-community/react-native-camera/issues/1530): *Could not find method google() for arguments [] on repository container #1530*

Within that issue, there it is [a solution that worked for my condition](https://github.com/react-native-community/react-native-camera/issues/1530#issuecomment-385752864). Because I am am new this the whole world of apps, and React Native, the above prompted me to ask the following questions which I appreciate if you can shed a light: 

* How the above situation may apply to developers, why does it or not, and why this is not part of any documentation? 

* Are there too many cases like this where manual patching is always necessary? It seems that the gradle world is one of such. If so, do you have references on articles or documentation that I can dive into that discussion to know more?

sibelius, 2018-07-08:
check this https://github.com/react-native-community/react-native-camera/blob/master/docs/GradleUpgradeGuide.md

feel free to create a script to make this manual steps not required
feel free to improve docs

taboca, 2018-07-09:
Thank you for pointing out this document. I see that this document is not linked from the readme and similarly the above referred issue 1530 was closed. 

Which makes me believe that maybe I have actually stumbled in a very special situation no? would you be able to help out with a few words about when such situation is relevant to general developer end-users that may try out the project?  

So far, in an attempt to understand the situation, I have simply produced an example [1], rawly created with react-create init, then patched it in conformity with your doc; it worked good.

[1] https://taboca.github.io/doc-js-example-qrcode-cam-example/

closed 2019-06-08 by stale[bot]