react-native-camera/react-native-camera #1686: Camera freezes and black screen

nadzic, 2018-07-09:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Come to one of screens where we initialize react-native camera
2. Scan the barcode with react-native camera
3. Camera freezes and we have to go next screen and back that we can scan again

### Does it work with Expo Camera?
Not using expo camera.

### Expected behaviour
React native camera should scan barcode normally and not freeze.

### Actual behaviour
I am using react native camera with the combination of the react-navigation 1.4.0, it works fine on iOS and newer version of Android phones, but on Android version 7.0 always I scan product barcode the camera freezes. If I want to scan, I have to go to next screen with react-navigation and then back to scan the product. 

I am having also another one issue, which happens sometimes - not every time. For example if I scan product or many products and go to next screen and then return back to this one screen I get black screen here where is initialised. It does not happen every time, just sometimes. I am using the following code for the scanning: 

`          <RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            style={styles.cameraPreview}
            onBarCodeRead={isEmpty(currentProduct) ? this.onBarCodeRead : null}
            permissionDialogTitle={'Permission to use camera'}
            permissionDialogMessage={'We need your permission to use your camera phone'}
          ></RNCamera>`

As you can see is am enabling/disabling onBarCodeRead with null or . Maybe that should be the problem? isEmpty(currentProduct) works fine, in cases when I need it is true, otherwise it is false.

### Environment
- **React Native version**: 0.52.0
- **React Native platform + platform version**: Android 7.0

### react-native-camera
**Version**: ^1.1.4

> Love react-native-camera? Please consider

Albert0405, 2018-07-09:
[react-native-qrcode-scanner](https://github.com/moaazsidat/react-native-qrcode-scanner) use this instead. It uses react-native-camera too (RNCamera)

nadzic, 2018-08-07:
Why should I use react-native-qrcode-scanner if I am scanning barcodes? What is actually the difference between react-native-camera and react-native-qrcode-scanner?

nadzic, 2018-09-13:
Is there any solution already?

Bartlebyy, 2018-10-08:
I had a similar problem. Looks like you're not unmounting the camera component when you switch between tabs. 
I recommend you take advantage of `componentDidMount` and added two listeners `willFocus` and `willBlur`.

Here is an example:
``` 
 componentDidMount() {
    const { navigation } = this.props;
    navigation.addListener('willFocus', () =>
      this.setState({ focusedScreen: true })
    );
    navigation.addListener('willBlur', () =>
      this.setState({ focusedScreen: false })
    );
  }

  render() {
    const { hasCameraPermission, focusedScreen } = this.state;
    if (hasCameraPermission === null) {
      return <View />;
    } else if (hasCameraPermission === false) {
      return <Text>No access to camera</Text>;
    } else if (focusedScreen){
      return (this.cameraView());
    } else {
      return <View />;
    }
  }
```

choo737, 2018-12-10:
for my case, these only works on second round of navigation. 
`navigation.addListener('willFocus', () =>
      this.setState({ focusedScreen: true })
    );
    navigation.addListener('willBlur', () =>
      this.setState({ focusedScreen: false })
    );`

"react-native-camera": "^1.4.3",
"react-native-qrcode-scanner": "^1.1.0",
"react-navigation": "2.18.3",
"react-navigation-redux-helpers": "^2.0.6",

n1ru4l, 2018-12-10:
@choo737 Could you please provide a minimal reproduction repository?

GaneshSaiKumar, 2019-01-11:
> [react-native-qrcode-scanner](https://github.com/moaazsidat/react-native-qrcode-scanner) use this instead. It uses react-native-camera too (RNCamera). If you're not using QRcode then you need to add barCodeTypes in the index.js of the plugin

i'm using react-native-qrcode-scanner still facing the same issue

paakcmd, 2019-03-04:
If you want to reuse the camera make sure it rerenders before the 2nd use.

superp, 2019-05-20:
> I had a similar problem. Looks like you're not unmounting the camera component when you switch between tabs.
> I recommend you take advantage of `componentDidMount` and added two listeners `willFocus` and `willBlur`.
> 
> Here is an example:
> 
> ```js
>  componentDidMount() {
>     const { navigation } = this.props;
>     navigation.addListener('willFocus', () =>
>       this.setState({ focusedScreen: true })
>     );
>     navigation.addListener('willBlur', () =>
>       this.setState({ focusedScreen: false })
>     );
>   }
> 
>   render() {
>     const { hasCameraPermission, focusedScreen } = this.state;
>     if (hasCameraPermission === null) {
>       return <View />;
>     } else if (hasCameraPermission === false) {
>       return <Text>No access to camera</Text>;
>     } else if (focusedScreen){
>       return (this.cameraView());
>     } else {
>       return <View />;
>     }
>   }
> ```

If you use Ignite and 'willFocus' doesn't fires try this https://github.com/react-navigation/redux-helpers/issues/53

n1ru4l, 2019-07-19:
@choo737 Do you have a reproduction?

wellington-cunha, 2019-08-16:
I solved my problem using  "react-native-camera": "1.2.0",

ntk0104, 2019-09-13:
> I had a similar problem. Looks like you're not unmounting the camera component when you switch between tabs.
> I recommend you take advantage of `componentDidMount` and added two listeners `willFocus` [...]

sorry but I have tried your solution and also tried with navigation.isFocused but none of them work? any solution?

closed 2019-11-19 by stale[bot]

JeevaS0722, 2020-01-18:
To make this work you need to:
```javascript
import { NavigationEvents } from 'react-navigation';
state = { loaded: true }
render() {
    const { loaded } = this.state;
    return (
      <View style={styles.container}>
        <NavigationEvents
          onWillFocus={payload => this.setState({loaded: true})}
          onDidBlur={payload => this.setState({loaded: false})}/>
      <View style={styles.cameraArea}>
        {loaded && (
          <Camera
            type={Camera.Constants.Type.back}
            ref={ref => {
              this.camera = ref;
            }}
          />
          )}

        </View>
```
The idea is to hide this camera view (onDidBlur-> loaded: false), then when you come back (onWillFocus is triggered and change loaded to true). When render() function is called it will show the <Camera /> again

https://stackoverflow.com/a/54308457/12029536

JayM96, 2020-04-11:
@JeevaS0722 thanks for the solution.