react-native-camera/react-native-camera #1687: Argument list too long: recursive header expansion failed at

zhangxinagjunHJH, 2018-07-09:
### 
Argument list too long: recursive header expansion failed at /Users/zxj/tfs2018/Huawei.GIS.Application/node_modules/react-native-camera/ios/../../../ios/HuaweiGIS/assets/web/bim-models/svf/cb214ab4-7bf0-41bd-a81c-2c81cd1d3f89.nwc/output.

### 

RNCamera

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?
Check usage with Expo - https://github.com/react-native-community/react-native-camera/blob/master/docs/Expo_Usage.md
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

Gys, 2018-07-14:
You could have searched the open and closed issues first. Because this same issue is opened 4 or 5 times already...

The solution is to modify the react-native-camera package every time that you run npm install. Because this problem is caused by a xcode project setting in the package. An explanation and workaround is here: https://github.com/react-native-community/react-native-camera/issues/1407#issuecomment-392881900

jgcmarins, 2018-07-16:
@Gys is this workaround the only solution or is there a final solution?

Gys, 2018-07-16:
The final solution is probably someone making a pull request with the updated xcode project ;-)

jgcmarins, 2018-07-16:
@Gys do you know how to do that?

Gys, 2018-07-16:
Have a look at this !
https://www.thinkful.com/learn/github-pull-request-tutorial/#Time-to-Submit-Your-First-PR

jgcmarins, 2018-07-16:
Hey @Gys,
I am very happy to know that you have good practices for submitting Pull Requests.
Could you help us on this one?
Since I've already done mine contributions: https://github.com/react-native-community/react-native-camera/commits?author=jgcmarins :wink:

dave-irvine, 2018-07-20:
@Gys @jgcmarins While we are discussing the issue, I noticed https://github.com/react-native-community/react-native-camera/issues/1454#issuecomment-406257449 that removing all entries from `FRAMEWORK_SEARCH_PATHS` also seems to resolve the issue. Does anyone know if this field needs to be filled at all?

sibelius, 2019-03-18:
check this https://github.com/react-native-community/react-native-camera/issues/1407

closed 2019-03-18 by sibelius